chipsalliance/Surelog #819: Potential bug in parse_model.tcl

hs-apotell, 2020-10-06:
The logic to populate ALL_CHILDREN list in `parse_model.tcl` seems buggy. Going by the name of the variable, its meant to be a map of class name to all its sub classes i.e. at every level in the hierarchy. But the logic doesn't seem to follow the intent.

For instance, the list of children for `expr` should include the following 47 classes -

```
bit_select
bit_var
byte_var
chandle_var
class_var
constant
enum_net
enum_var
func_call
indexed_part_select
int_var
integer_net
integer_var
let_expr
logic_net
logic_var
long_int_var
method_func_call
method_task_call
net
net_bit
nets
operation
packed_array_net
packed_array_var
parameter
part_select
real_var
ref_obj
short_int_var
short_real_var
simple_expr
spec_param
string_var
struct_net
struct_var
sys_func_call
sys_task_call
task_call
tf_call
time_net
time_var
union_var
unsupported_expr
var_bit
var_select
variables
```

but it reports only the following 21 classes -

```
bit_select
constant
func_call
indexed_part_select
let_expr
method_func_call
method_task_call
nets
operation
parameter
part_select
ref_obj
simple_expr
spec_param
sys_func_call
sys_task_call
task_call
tf_call
unsupported_expr
var_select
variables
```

Disclaimer: I might as well be reading it wrong or this might be intentional. My knowledge of tcl is rudimentary at best.

alaindargelas, 2020-10-07:
Yup that is a pretty major bug!
For some reason it was not obviously apparent at runtime...
It impacts the groups runtime checks.

alaindargelas, 2020-10-07:
Fixed by https://github.com/alainmarcel/UHDM/pull/355

closed 2020-10-07 by alaindargelas