chipsalliance/Surelog #822: Model files in UHDM fail YAML validation check

hs-apotell, 2020-10-07:
Look like none of the files are legal yaml format. Intentional?

None of the publicly available yaml parsers can be be used to parse these files. :(

alaindargelas, 2020-10-07:
Correct, it's yaml like.
Simplified to the bare minimum to minimize typing.
What are you trying to achieve?

hs-apotell, 2020-10-08:
> Simplified to the bare minimum to minimize typing.

Seems counter productive since UHDM is meant to be a bridge, a foundational layer that you build on. Hope the deviation from the standard is worth it in the long run.

> What are you trying to achieve?

Something a little ambitious. Attempting to see if I can use [hdlConvertor](https://github.com/Nic30/hdlConvertor) project with UHDM. Basically, comparing and contrasting the outputs of two independent parsers.

closed 2020-10-08 by hs-apotell

mithro, 2020-10-08:
@alaindargelas - Why wouldn't you just use yaml?

alaindargelas, 2020-10-09:
The model format is internal and has no bearing on the actual generated code (the public UHDM C++ code).I didn't want to use a complex yaml parser and heavier syntax to capture the model.My Yaml like parser is less than 100 lines of code, including creating the in-memory data model for it.I am capturing the model by myself basically, path of lesser resistance would be my explanation. The model is now 90% complete for our needs, and the syntax I picked allows for all aspects of the modeling.There is no template generator out there that actually generates the kind of C++ implementation we need, so using a standard input language has no value since the whole code generation had to be created from scratch. I just needed something that is human readable and Yaml was a good source of inspiration. It does not have to be Yaml exact.

alaindargelas, 2020-10-09:
Simplified to the bare minimum to minimize typing.

Seems counter productive since UHDM is meant to be a bridge, a foundational layer that you build on. Hope the deviation from the standard is worth it in the long run.

>> The "bridge" offered by UHDM is the generated code, the C++ actual implementation, not the text input.

alaindargelas, 2020-10-09:
The Yaml deviation is of no zero concern here.
What matters is the generated code. UHDM is not about the capturing of the SystemVerilog Object Model in language x,y,z, it is about generating an actual implementation of Classes and Utilities in C++ and make use of the generated implementation in many applications.