chipsalliance/Surelog #826: Function related callbacks in visitor logic

hs-apotell, 2020-10-10:
Here's a test scenario, two functions calling each other recursively -
```
module Recursion;
  int x,y,z,a;

  function test1(int z);
    test2(z + 1);
  endfunction

  function test2(int z);
    test1(z + 1);
  endfunction 

  initial begin
    x = 10;
    y = 20;
    test1(x);
  end
endmodule
```
And, here's the output of 4 functions related callbacks - 

'''
Entering Module Instance: work@Recursion
enterFunc_call: line 15
  enterFunction: line 4
   enterFunc_call: line 5
   leaveFunc_call: line 5
  leaveFunction: line 4
leaveFunc_call: line 15

enterFunc_call: line 5  <= This should have followed with enterFunction: line 8
leaveFunc_call: line 5

enterFunc_call: line 9
  enterFunction: line 4  <= Or optionally, this shouldn't have been here!
   enterFunc_call: line 5
   leaveFunc_call: line 5
  leaveFunction: line 4
leaveFunc_call: line 9
Leaving Module Instance: work@Recursion
'''

There seems to be some inconsistency in these calls. I can't seem to be find a consistent pattern on when to expect a enterFunction callback.

Also, should the enterFunc_call even walk into enterFunction? If yes, what happens in case of recursions?

Thomasb81, 2020-10-10:
What do you expect from such testcase ? 
It will crash at simulation.

What the trace is supposed to represent ?

alaindargelas, 2020-10-11:
I think you are confusing the listener with an interpreter/simulator. 
The listener does not follow any order of execution, it simply traverses the UHDM data model, and the order of traversal is not following a standard or a strict semantic. It only guaranties that all relations get traversed.

Details:

- Make sure you have the -elabuhdm option in the Surelog command line
  Or that in your client code you invoke the uhdm Elaborator before you listen with your listener:
```
   ElaboratorListener* listener = new ElaboratorListener(&serializer, false);
    listen_designs(restoredDesigns, listener);
```

- Check the UHDM/src/vpi_listener.cpp file, you'll see the order taken to listen to all event.
  - First the listener listens to the non-elaborated tree (allModules)
  - Then it listens to the elaborated tree (topModules)

The order the functions get traversed is not representative of an "execution" order, the listener is not a simulator.
It follows the order in the models.yaml files. Once a relation is traversed, it is not traversed again. Non-elaborated and elaborated relations are not equivalent, so you'll see both.

The listener does not "loop" as it does not execute the code, it only traverses one time the relations.
func_calls do not walk into function body, that would be some form of execution which is not the goal of the listener.

You should be able to make sense of the traversal if you step in the debugger and inspect the vpi_listener.cpp code. You'll see the order follows the declarations in the models.yaml files (But that does not have any particular meaning either).

alaindargelas, 2020-10-11:
It looks like you need to use the existing listener to create your own data structure, then on that custom datastructure, apply the "execution" order you are looking for (Sounds like an interpreter).

Or write a code generator like the one that generates the listener and add it your own semantic for order and traversal whatever that is.

hs-apotell, 2020-10-11:
Sorry about the bad example. I just happen to be working with something more complex but with similar pattern so happens to be on my mind. Let me try again with simpler (and wouldn't _crash_ the simulator) -

```
1  module Test;
 2   int x = 10, y = 20, z;
 3
 4   initial begin
 5     test(x, y);
 6     x = 50;
 7     test(x, y);
 8   end
 9
10   function test(int a, b);
11     z = a + b;
12   endfunction
13 endmodule
```
And here's the listener output. I have 6 functions implemented in a subclass of VpiListener i.e. enter/leave for func_call, function, and assignment.
```
enterFunc_call: line 5
-- enterFunction: line 10
-- enterAssignment: line 11
-- leaveAssignment: line 11
-- leaveFunction: line 10
leaveFunc_call: line 5
enterAssignment: line 6
leaveAssignment: line 6
enterFunc_call: line 7
-- enterFunction: line 10
-- enterAssignment: line 11
-- leaveAssignment: line 11
-- leaveFunction: line 10
leaveFunc_call: line 7
enterAssignment: line 11
leaveAssignment: line 11
```
I don't expect line prefixed with '--' above in that order. As per your comment above -

> The listener does not follow any order of execution
> Once a relation is traversed, it is not traversed again.
> The listener does not "loop" as it does not execute the code, it only traverses one time the relations.
> func_calls do not walk into function body, that would be some form of execution which is not the goal of the listener.

But that is exactly what's happening above. Listener is following the order of execution. The assignment statement on line 11 was traversed twice because of the function getting called twice and each time it actually did step into the function and reported the assignment.

Here's what I would expect as the output -
```
enterFunc_call: line 5
leaveFunc_call: line 5
enterAssignment: line 6
leaveAssignment: line 6
enterFunc_call: line 7
leaveFunc_call: line 7
enterFunction: line 10
enterAssignment: line 11
leaveAssignment: line 11
leaveFunction: line 10
```

Since, the listener is NOT a interpreter/simulator or like one, the context should NOT jump into the function when it sees a function call. That is the job of the _simulator-like_ client of UHDM to decide.

alaindargelas, 2020-10-11:
Sorry, I completely forgot a few details on the listener since I implemented it:

1) There is nothing preventing repeats right now in the vpi_listener, 
    Unlike the vpi_visitor.cpp which prevents re-printing the same object twice. Note that makes the vpi_visitor an incomplete walker, but it's intend is to "print" the model, not to be used by subsequent clients.

    Multiple calls like in a seq_block will indeed traverse the object multiple times.
    The client code needs to know there are multiple object calls (func call in this case), 
    It should be up to the client to decide to 
    ignore traversing the recursive relations multiple times (Use object IDs to detect object are already visited) if need be.

2) The traversal dictated by the model:

From: UHDM/src/vpi_listener.cpp
```
...
void UHDM::listen_func_call(vpiHandle object, VpiListener* listener) {
  func_call* d = (func_call*) ((const uhdm_handle*)object)->object;
  const BaseClass* parent = d->VpiParent();
  vpiHandle parent_h = parent ? NewVpiHandle(parent) : 0;
  listener->enterFunc_call(d, parent, object, parent_h);
    vpiHandle itr;
    itr = vpi_handle(vpiFunction,object);
    if (itr) {
      listen_function (itr, listener);
      vpi_free_object(itr);
    }
```

The func_call does indeed listen to the function declaration as this relation exist in the model (It is the only relation as a matter of fact):

From: UHDM/model/func_call.yaml

```
- obj_def: func_call
  - extends: tf_call
...
  - obj_ref: function
    name: function
    type: function
    vpi: vpiFunction
    card: 1
```

3) Your execution trace matches the description above:

```
enterFunc_call: line 5
-- enterFunction: line 10
-- enterAssignment: line 11
-- leaveAssignment: line 11
-- leaveFunction: line 10
leaveFunc_call: line 5
enterAssignment: line 6
leaveAssignment: line 6
enterFunc_call: line 7
-- enterFunction: line 10
-- enterAssignment: line 11
-- leaveAssignment: line 11
-- leaveFunction: line 10
leaveFunc_call: line 7
```

4) Except for the last 2 lines:
```
enterAssignment: line 11
leaveAssignment: line 11
```
You need to check the stack in gdb to explain why you get there.
The fact that you are not listening to many events in your trace makes it hard to "debug" based on the text output.

5) This desired trace of yours traverses the tree and skips arbitrary relations (Like it skips the vpiFunction above).
It is only interested in special events too it looks like.
This is not something that can be generated automatically, 
 - you need to write full custom code, 
 - or auto generate your version of vpi_listener.cpp based on a locally edited model?
 - or use the listener as is, listen and build your data structure with the objects of interest, then do you own traversal in the order you wish.

```
enterFunc_call: line 5
leaveFunc_call: line 5
enterAssignment: line 6
leaveAssignment: line 6
enterFunc_call: line 7
leaveFunc_call: line 7
enterFunction: line 10
enterAssignment: line 11
leaveAssignment: line 11
leaveFunction: line 10
```

hs-apotell, 2020-10-12:
Full trace log [FullTrace.log](https://github.com/alainmarcel/Surelog/files/5362945/FullTrace.log) of the above shared example attached for reference.

> Except for the last 2 lines:

Those lines are printed because of stepping into ```enterTask_func```

> This desired trace of yours traverses the tree and skips arbitrary relations

Its expectation of consistency. If the logic skips once it should skip always and vice-versa. By no means is this use case arbitrary. When _similar_ looking statements are encountered, _similar things_ happen. Which, unfortunately, doesn't seem to be the case when dealing with functions. More information below about other use cases that also produce different/unexpected output.

This proposed/designed logic would create infinite loops with listener which will be out of control of the client code. A simple example -
```
module Infinite;
  int x = 10, y, z;

  initial begin
    y = test(x);
  end

  function test(int a);
    if (a <= 1)
      return 1;
    z = test( a - 1);
    return z;
  endfunction
endmodule
```
The above is a infinite loop which the client side cannot do anything about. Note that a simulator would do just fine with such a program since it evaluates the precondition on top of the function. VpiListener isn't evaluating such preconditions that would prevent the function from executing to the end. That's the job of _simulator like_ applications/clients.

Also, another potential bug - statements like ```return test(a-1);``` neither actually enter the func_call nor enter the function. Trace log [ReturnFunction.log](https://github.com/alainmarcel/Surelog/files/5362938/ReturnFunction.log) of this scenario attached.

Function calls within functions do not enter/leave function. Here's a test case and trace log. [FuncCallsFunc.log](https://github.com/alainmarcel/Surelog/files/5362956/FuncCallsFunc.log). This behavior is different from when the function is being called from the initial block.

```
module Test;
  int x = 10, y = 20, z;

  initial begin
    testA(x, y);
  end

  function testA(int a, b);
    testB(a, b);
  endfunction

  function testB(int a, b);
    testC(a, b);
  endfunction

  function testC(int a, b);
    z = a + b;
  endfunction
endmodule
```
Functions needs to be treated *specially*. They change program context which AFAIK no other programming construct does (does SV have goto, jump and long jump concepts?). Listener logic shouldn't step into the function implicitly like every other relation does. Client has no way to this behavior and bookkeeping of all the different scenarios of when such callbacks are valid isn't practical. On the other hand, in the absence of ```enterFunc_call``` not stepping into ```enterFunction```, if the client need to step into the function, he/she can always explicitly call ```listen_function``` from ```enterFunc_call``` which gives full control of the flow. This change in behavior will also ensure that the trace always runs from top to bottom in increasing line order and not jump around like it does now - much more predictable (and expected, in my opinion) behavior.

alaindargelas, 2020-10-12:
Please propose a changelist in UHDM/model_gen/model_gen.tcl with your new proposed behavior for the vpi_listener.

hs-apotell, 2020-10-12:
New PR - https://github.com/alainmarcel/UHDM/pull/357

And here are the revised logs with this change -
[FuncCallsFunc.log](https://github.com/alainmarcel/Surelog/files/5364583/FuncCallsFunc.log)
[Infinite.log](https://github.com/alainmarcel/Surelog/files/5364584/Infinite.log)
[FullTrace.log](https://github.com/alainmarcel/Surelog/files/5364585/FullTrace.log)

alaindargelas, 2020-10-13:
OK, I updated the logs.

closed 2020-10-13 by alaindargelas