chipsalliance/Surelog #828: UHDM::visit_object doesn't handle vpiProcess

hs-apotell, 2020-10-11:
```UHDM::listen_process_stmt``` iterates ```vpiAttribute``` and ```vpiStmt``` members but there is no such corresponding handling of the same members in ```UHDM::visit_object```.

Following snippet in missing as part of auto-generated code.

```
  if (objectType == vpiProcess) {
    itr = vpi_iterate(vpiAttribute, obj_h);
    while (vpiHandle obj = vpi_scan(itr)) {
      visit_object(obj, subobject_indent, "vpiAttribute", visited, out);
      release_handle(obj);
    }
    release_handle(itr);
    itr = vpi_handle(vpiStmt, obj_h);
    visit_object(itr, subobject_indent, "vpiStmt", visited, out);
    release_handle(itr);
    return;
  }
```

alaindargelas, 2020-10-11:
That is because "process" is a virtual class in the model, "always" and "initial" are the concrete objects. The walker only traverses concrete object.

From: UHDM/src/vpi_visitor.cpp

``` 
 if (objectType == vpiAlways) {
    if (const int n = vpi_get(vpiAlwaysType, obj_h))
      if (n != -1)
        stream_indent(out, indent) << "|vpiAlwaysType:" << n << "\n";

    vpiHandle itr;
    itr = vpi_iterate(vpiAttribute,obj_h);
    while (vpiHandle obj = vpi_scan(itr) ) {
      visit_object(obj, subobject_indent, "vpiAttribute", visited, out);
      release_handle(obj);
    }
    release_handle(itr);
    itr = vpi_handle(vpiStmt,obj_h);
    visit_object(itr, subobject_indent, "vpiStmt", visited, out);
    release_handle(itr);

    return;
  }
```

alaindargelas, 2020-10-12:
Is there a reason for reopening this issue?

hs-apotell, 2020-10-12:
> Is there a reason for reopening this issue?

Yes, I started commenting and then got distracted. Needed some clarification for better understanding.

> "process" is a virtual class

Did you mean _abstract_ and not _virtual_?

Assuming you meant _abstract_ class, there are other abstract classes like ```disables```, ```expr```, ```net``` and many others that do get included as part of ```UHDM::visit_object```. What makes ```process``` different?

Also, since ```process``` is abstract, ```any::UhdmType()``` implementation will ever return ```UHDM::uhdmprocess_stmt```? So, the case block in ```UHDM::listen_any``` for ```uhdmprocess_stmt``` is dead code? Or, can a ```uhdm_handle``` hold an object of concrete type and still have it's type member set to one its super classes? For example, would a ```uhdm_handle``` with type ```uhdmatomic_stmt``` and a pointer to ```disables``` be valid? I raise this because ```UHDM::listen_any``` looks at the handle's ```type``` member and not at ```object->UhdmType()``` - both of which could be different.

alaindargelas, 2020-10-12:
Let's stick to the Verilog Object Model lingo:
A "process" is a "class", (So you can call it abstract or virtual)
An "always" is an "object".

UhdmType() should always return a concrete object type, so no it won't return UHDM::uhdmprocess_stmt.
Correct, all the switch case in listen_any for classes is dead code.
All types (vpiType) have a corresponding UhdmType.
A uhdm_handle always have an object type, not a class type.
class types are only used in relation traversal (per the model).

hs-apotell, 2020-10-12:
Appreciate the detailed explanation. That helps a lot.

I am studying the model_gen.tcl code to get a better understanding of the code-generation process. There is some dead code in both VpiListener and vpi_visitor implementations. May be I can create a PR as a "good first issue".

closed 2020-10-12 by hs-apotell