chipsalliance/Surelog #836: UVM/OVM warnings and errors

hs-apotell, 2020-10-13:
There are a few warnings/errors reported when running the UVM/OVM jobs while building. They happen only on Windows. Throwing it out here to see if either of you have any clue/direction on identifying the root cause.

```
"       Creating OVM precompiled package..." 
  File "<string>", line 1
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape
"       Package OVM created"
[475/477] Generating bin/PrecompileUVM.timestamp
"       Creating UVM precompiled package..." 
  File "<string>", line 1
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape
"       Package UVM created"
```

Here's a link to the relevant build. https://travis-ci.com/github/anadishukla/Surelog/jobs/398681489

Also, incremental builds generate a different error which is kind of odd.
```
"       Creating OVM precompiled package..." 
  <string>:1: DeprecationWarning: invalid escape sequence \P
  "       Package OVM created"
  [76/77] cmd.exe /C "cd /D XXXXX\build\debug\bin && echo "       Creating UVM precompiled package..." && XXXXX\build\debug\bin\surelog -nobuiltin -createcache +incdir+.+1800.2-2017-1.0/src/ 1800.2-2017-1.0/src/uvm_pkg.sv -writepp -mt 0 -parse -nocomp -noelab -nostdout && echo "       Package UVM created""
  "       Creating UVM precompiled package..." 
  <string>:1: DeprecationWarning: invalid escape sequence \P
  "       Package UVM created"
```

A online search to the relevant error text points in the direction of some wrong escape sequence in python. I haven been unable to locate any such thing in full text code search.

We suspect that this issue is root cause of some of the regression test failures on Windows. We haven't been able to confirm successfully yet though.

Appreciate any help I can get in identifying the root cause of this issue.

alaindargelas, 2020-10-14:
I'd play with the unix2dos utility and apply it to the Python files in Surelog:
ls */*/*.py
src/API/python_listener.py         src/API/vobjecttypes.py                    tests/UnitClass/my_listener.py         tests/UnitPython/myscriptPerFile.py
src/API/slapi.py                   tests/SimpleClass1/my_listener.py          tests/UnitClass/myscriptPerDesign.py   third_party/flatbuffers/conanfile.py
src/API/slformatmsg.py             tests/SimpleClass1/myscriptPerDesign.py    tests/UnitClass/myscriptPerFile.py
src/API/slSV3_1aPythonListener.py  tests/SimpleClass1/myscriptPerFile.py      tests/UnitPython/my_listener.py
src/API/slwaivers.py               tests/SimpleIncludeAndMacros/slwaivers.py  tests/UnitPython/myscriptPerDesign.py

Something in one of these files contains that illegal sequence.

alaindargelas, 2020-10-14:
Maybe one of these?

grep  '\\' src/API/*.py
src/API/python_listener.py:SLregisterNewErrorType("[NOTE :PY0403]", "Module declaration \"%s\"", "");
src/API/slformatmsg.py:    chunks.append("\n\n")
src/API/vobjecttypes.py:"# This file is automatically generated by generate_parser_listener.tcl\n"
src/API/vobjecttypes.py:"# DO NOT EDIT\n"
alain@alain-mac:~/Surelog$

alaindargelas, 2020-10-14:
What happens if you add the -nopython to the Surelog command line for the failing tests?

hs-apotell, 2020-10-14:
I narrowed down the root cause to ```m_programPath``` variable in PythonAPI.cpp. I have an open PR #837 with the relevant change to fix the issue.

Regardless, it might still be worthwhile to do a quick pass of unix2dos utility to find any other similar problems.

Unfortunately, that didn't fix the errors we are seeing in the tests. We have about 14 tests failing on Windows which report success on Unix. We are in the process of summarizing all the failures and identifying the causes. If need be I will open a follow up issue for input.

alaindargelas, 2020-10-15:
The IEEE Surelog/third_party/UVM/1800.2-2017-1.0/ itself could have an issue in Windows (Files containing illegal characters) ?

hs-apotell, 2020-10-15:
Looks like the errors are unrelated to this. Closing this issue and continuing the discussion on Windows CI issue.

closed 2020-10-15 by hs-apotell