chipsalliance/Surelog #844: Windows build: Development Environment Required

DamienLeBars, 2020-10-16:
Hello,
I have issues trying to compile Surelog on Windows(10) with make
**my env:**
GNU Make 4.2.1 (Built for i686-pc-cygwin)
cmake version 3.18.2
visual studio 2017

1. TCMALLOC_LIBRARY
...
```...
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
TCMALLOC_LIBRARY
    linked by target "surelog" in directory C:/git/Surelog

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:20: run-cmake-release] Error 1
```

resolved adding condition
if(TCMALLOC_LIBRARY)
for target_link_libraries calls with ${TCMALLOC_LIBRARY}  in CMakeLists.txt files

2. *** No targets specified and no makefile found.  Stop.
```...
make -C build
make[1]: Entering directory '/cygdrive/c/git/Surelog/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/cygdrive/c/git/Surelog/build'
make: *** [Makefile:13: release] Error 2
```
Is it possible (perhaps @hs-apotell ) to update the Development Environment Required section of [INSTALL.mdl](https://github.com/alainmarcel/Surelog/blob/master/INSTALL.mdl) file for windows?

DamienLeBars, 2020-10-16:
i have switched to full cygwin env but i have this error:
/third_party/UHDM/src/Serializer_restore.cpp:13296:3: error: ‘close’ was not declared in this scope; did you mean ‘pclose’?
13296 |   close(fileid);
      |   ^~~~~~
      |   pclose
maybe the use of  POSIX functions open/close can be changed to standard C fopen/fclose?

hs-apotell, 2020-10-16:
Here's my environment settings and the build succeeds, at least locally on more than one machine.

> update the Development Environment Required section

Yes. It's on my TODO. I was just hoping to set up CI before I announce official Windows support.

P.S. This is Visual Studio environment settings i.e. CMakeSettings.json used by Visual Studio. You can safely translate these into command line as well.

```
{
  "environments": [
    {
      "environment": "msvc_x64_x64",
      "PATH": "${env.PATH};D:\\SysinternalsSuite;D:\\tclkit\\8.7a3-1\\bin"
    }
  ],
  "configurations": [
    {
      "name": "x64-Release",
      "generator": "Ninja",
      "configurationType": "RelWithDebInfo",
      "inheritEnvironments": [ "msvc_x64_x64" ],
      "buildRoot": "${projectDir}\\CMakeBuilds\\build\\release",
      "installRoot": "${projectDir}\\CMakeBuilds\\install\\release",
      "cmakeCommandArgs": "-DCMAKE_POLICY_DEFAULT_CMP0091=NEW",
      "buildCommandArgs": "-v",
      "ctestCommandArgs": "",
      "variables": [
        {
          "name": "SWIG_DIR",
          "value": "D:/swig/3.0.12/Lib",
          "type": "PATH"
        },
        {
          "name": "SWIG_EXECUTABLE",
          "value": "D:/swig/3.0.12/swig.exe",
          "type": "FILEPATH"
        },
        {
          "name": "NO_TCMALLOC",
          "value": "ON",
          "type": "BOOL"
        },
        {
          "name": "JAVA_HOME",
          "value": "D:\\Java\\jre-1.8.0_241",
          "type": "PATH"
        },
        {
          "name": "FLATBUFFERS_INSTALL",
          "value": "False",
          "type": "BOOL"
        },
        {
          "name": "INSTALL_GTEST",
          "value": "False",
          "type": "BOOL"
        }
      ]
    },
    {
      "name": "x64-Debug",
      "generator": "Ninja",
      "configurationType": "Debug",
      "inheritEnvironments": [ "msvc_x64_x64" ],
      "buildRoot": "${projectDir}\\CMakeBuilds\\build\\debug",
      "installRoot": "${projectDir}\\CMakeBuilds\\install\\debug",
      "cmakeCommandArgs": "-DCMAKE_POLICY_DEFAULT_CMP0091=NEW",
      "buildCommandArgs": "-v",
      "ctestCommandArgs": "",
      "variables": [
        {
          "name": "SWIG_DIR",
          "value": "D:/swig/3.0.12/Lib",
          "type": "PATH"
        },
        {
          "name": "SWIG_EXECUTABLE",
          "value": "D:/swig/3.0.12/swig.exe",
          "type": "FILEPATH"
        },
        {
          "name": "NO_TCMALLOC",
          "value": "ON",
          "type": "BOOL"
        },
        {
          "name": "JAVA_HOME",
          "value": "D:\\Java\\jre-1.8.0_241",
          "type": "PATH"
        },
        {
          "name": "FLATBUFFERS_INSTALL",
          "value": "False",
          "type": "BOOL"
        },
        {
          "name": "INSTALL_GTEST",
          "value": "False",
          "type": "BOOL"
        }
      ],
      "cmakeExecutable": "D:/cmake/3.15.4/bin/cmake.exe"
    }
  ]
}
```

DamienLeBars, 2020-10-16:
Ok, thanks a lot for your fast answer and your efforts. I will try it.

DamienLeBars, 2020-10-27:
@hs-apotell, great job!  o// 
Using your windows-msvc config, I succeeded locally compile and run tests on my computer (with visual studio 2017).

Note 1: I had to add "#include <cctype>" due to std::isalpha function in src/SourceCompile/PreprocessFile.cpp line 584 to avoid compilation error.

Note 2: Some regressions tests fails (ClassCons, DiffSimpleIncludeAndMacros)

hs-apotell, 2020-10-28:
> I had to add "#include " due to std::isalpha function in src/SourceCompile/PreprocessFile.cpp line 584 to avoid compilation error.

This must be VS2017 specific compiler issue. I have only tested with the latest VS2019 compiler. There is no official minimum requirement stated but I would recommend using VS2019.

> Some regressions tests fails (ClassCons, DiffSimpleIncludeAndMacros), the problem come from '*' in argument file/directory.

Yes I am aware of this discrepancy. The same tests work fine on the CI machines but locally they fail. I haven't exactly narrowed down the root cause of it but I know it has to do with the wildcard arguments passed down to the executable. For whatever reason, the arguments expand (like "*.sv" would expand to "a.sv b.sv", just like on bash shell) on the CI machines but they don't on the local command window. AFAIK Windows Command shell doesn't do this but the CI machines are contradicting. Need some time to investigate this.

DamienLeBars, 2020-10-28:
ok,
a little investigation orient me here:
[https://docs.microsoft.com/en-us/cpp/c-language/expanding-wildcard-arguments?view=vs-2019](url)
So, a solution is to add this in CMakeLists.txt (around line 374):
set_target_properties(surelog-bin PROPERTIES LINK_FLAGS "setargv.obj")
potential consequence: the  NOTE-UHDM count seems to be increased in some tests...

hs-apotell, 2020-10-28:
That's an interesting find. We aren't linking the necessary file and yet we are seeing difference on CI vs. local.

Interestingly, it was my understanding that the wildcard => expanded args is a function of the shell. How does linking an object file for a specific application change that behavior? Something very murky going on here.

alaindargelas, 2020-11-06:
The Github actions on Windows should provide complete documentation. Can we close this?

DamienLeBars, 2020-11-06:
For my initial problem (get a windows compilation ok), this is no longer a priority. Thanks again @hs-apotell .

closed 2020-11-30 by alaindargelas