chipsalliance/Surelog #864: Ranges missing in parameter declarations

rkapuscik, 2020-10-19:
For this test: https://github.com/alainmarcel/uhdm-integration/blob/master/tests/OneInside/dut.v#L5
Created UHDM:
```
|vpiParamAssign:
  \_param_assign: , line:5, parent:work@dut
    |vpiRhs:
    \_constant: , line:5
      |vpiConstType:3
      |vpiDecompile:3'b100
      |vpiSize:3
      |BIN:3'b100
    |vpiLhs:
    \_parameter: (work@dut.OP_1), line:5, parent:work@dut
      |vpiName:OP_1
      |vpiFullName:work@dut.OP_1
      |vpiLocalParam:1
      |vpiTypespec:
      \_bit_typespec: , line:5, parent:work@dut.OP_1
```
and further on:
```
  |vpiParameter:
  \_parameter: (work@dut.OP_1), line:5, parent:work@dut   
    |vpiName:OP_1
    |vpiFullName:work@dut.OP_1
    |INT:4    
```
There is no information regarding width of the parameter.
On the other hand, Verilator does store both the range and the value:
```
1:2: VAR 0x55d81a0cc580 <e94#> {c5at} @dt=0@  OP_1 LPARAM
    1:2:1: BASICDTYPE 0x55d81a148c30 <e92#> {c5an} @dt=this@(nw0)  logic kwd=logic
    1:2:1:1: RANGE 0x55d81a14c3c0 <e85#> {c5an}
    1:2:1:1:2: CONST 0x55d81a144c60 <e77#> {c5ao} @dt=0x55d81a148000@(G/swu32/2)  ?32?sh2
    1:2:1:1:3: CONST 0x55d81a144d80 <e78#> {c5aq} @dt=0x55d81a1480d0@(G/swu32/1)  ?32?sh0
    1:2:3: CONST 0x55d81a144b40 <e93#> {c5ba} @dt=0x55d81a148b60@(G/w3)  3'h4
```

alaindargelas, 2020-10-20:
Can you propose a PR? It should be a simple problem in compileTypespec range computation call due to implicit type.
I'm knee deep in class support.

alaindargelas, 2020-10-24:
Fixed by #869

closed 2020-10-24 by alaindargelas