chipsalliance/Surelog #874: [Windows Only] Wildcard expansion on command line

hs-apotell, 2020-10-28:
We have identified some discrepancy in the way the arguments expand for regression tests. These tests run successfully on the CI machines but they fail locally. Here's one example for the test named `ClassCons`.

The input to the test from `ClassCons.sl`,
```
-timescale=1ns/1ns +vcs+flush+all  +warn=all -sverilog   -d ast -d inst +incdir+.+../../../UVM/uvm-1.2/src/  -writepp    -verbose  -mt max -parse -fileunit *.pkg  top.sv
```
On the CI machines, from `slpall/surelog.log` file
```
-timescale=1ns/1ns +vcs+flush+all +warn=all -sverilog -d ast -d inst +incdir+.+../../third_party/UVM/uvm-1.2/src/ -writepp -verbose -parse -fileunit definesPkg.pkg definesPkg_1.pkg top.sv -mt 0 -mp 0 -o ../../build/tests/ClassCons/
```
On local machines,  from `slpall/surelog.log` file
```
-timescale=1ns/1ns +vcs+flush+all  +warn=all -sverilog   -d ast -d inst +incdir+.+../../third_party/UVM/uvm-1.2/src/  -writepp    -verbose   -parse -fileunit *.pkg  top.sv  -mt 0 -mp 0 -o ../../build/tests/ClassCons/
```
Notice how `*.pkg` in the argument got expanded to `definesPkg.pkg definesPkg_1.pkg` on the CI machines but not on the local run. Both CI and local are _expected_ to use the `cmd.exe` to run the test.

AFAIK `cmd.exe` doesn't do this sort of expansion. So what in the process is doing this? How do I repeat what CI machines are doing?

hs-apotell, 2020-10-28:
@umarcor Do you know anything specific about the GHA Windows machines that would be responsible for this behavior?

umarcor, 2020-10-28:
@hs-apotell would you mind pointing to the line in `ClassCons.sl` where that command is generated? Typically these issues are produced because the shell interprets the wildcard before passing it to the command. But it is difficult to check it without further context.

hs-apotell, 2020-10-28:
The file is in the repository at `tests/ClassCons/ClassCons.sl`

Here's a build where I attempted to dump the logs for failed ones to see what the CI node was doing.
https://github.com/hs-apotell/Surelog/runs/1305736255?check_suite_focus=true

And attached is the log file from local run.
[ClassCons.log](https://github.com/alainmarcel/Surelog/files/5449894/ClassCons.log)

umarcor, 2020-10-28:
It seems that https://github.com/alainmarcel/Surelog/blob/master/tests/ClassCons/ClassCons.sl holds certain arguments that some script/source reads and passes to Surelog. Which is that script/source?

hs-apotell, 2020-10-28:
> Which is that script/source?

`tests/regression.tcl` Note that this is the same script running on both CI and local. No local changes.

Here's the complete regression test flow -
1. Run `tests/cmake_gen.tcl` which generates `build/test/CMakeLists.txt`. Each test is an independent custom cmake command.
2. Use cmake to build and run the generated CMakeLists.txt file.
3. All test output goes in `build/tests/<test-name>` folder

umarcor, 2020-10-28:
I would suggest checking how is https://github.com/alainmarcel/Surelog/blob/master/tests/regression.tcl#L123-L130 being evaluated both locally and in CI.

hs-apotell, 2020-10-28:
Why would those lines change anything? Unless tcl_platform returns a different value between CI and local, I don't see how this will change anything. To be clear, we are _explicitly_ [calling cmd](https://github.com/alainmarcel/Surelog/blob/3b8347b5678fbff7ebf467b6f88f6d810aefe170/tests/regression.tcl#L460) from the script (and not leaving up to the parent shell or cmake to make that decision).

A few things I have tried -
* Calling the surelog.exe directly without wrapping it with cmd shell (This, unfortunately, isn't recommended in TCL)
* Using the `%ComSpec% /D /E:ON /V:OFF /S /C "CALL "{0}""` as stated [here](https://docs.github.com/en/free-pro-team@latest/actions/reference/workflow-syntax-for-github-actions#using-a-specific-shell)

Neither of these made a difference locally.

umarcor, 2020-10-28:
> Why would those lines change anything? Unless tcl_platform returns a different value between CI and local

I would expect `tcl_platform` to return the same value, but I'm concerned with MSYSTEM. Since tclsh from https://github.com/alainmarcel/Surelog/blob/master/.github/workflows/main.yml#L195 is used (which is an unsupported use case), I'm not sure about which environment is the tcl shell seeing. The tcl shell is certainly part of an MSYS/MINGW installation. Even though it should be usable as an independent binary outside, the build might be customised.

Just to be sure, you don't want `*` to be expanded, do you?

hs-apotell, 2020-10-28:
> you don't want * to be expanded, do you?

No, I don't want it to be expanded in runs from CMD/COMSPEC because that's not a default platform behavior. If existence of MSYS/MinGW on the machine makes a difference, that's a problem. I would expect the build to behave identical with or without MSYS/MinGW in picture.

umarcor, 2020-10-28:
The existence of MSYS2/MINGW is not a problem, per se. But those are added to the PATH, and you are adding Git for Windows' path explicitly. Again, it should not have any effect, but who knows.

alaindargelas, 2020-12-29:
One easy fix is to actually expand the *.v commands to the actual files they represent in the .sl files.
Can we do just that and close this issue? 
Having *.v as a file list is not best practice anyhow.

hs-apotell, 2020-12-29:
> One easy fix is to actually expand the *.v commands to the actual files they represent in the .sl files.

I agree. However, if possible, we should understand the current behavior. Why is Windows CI build succeeding when it should fail? More importantly the CI build succeeds but locally it doesn't. Having this difference between local and CI is problematic.