chipsalliance/Surelog #892: Ranges are missing in unused (and not top) module ports

kamilrakoczy, 2020-11-10:
Test: https://github.com/antmicro/uhdm-integration/blob/unused-module/tests/UnusedModule/top.sv
There is no information regarding width of the ports in unused (and not top) module.

Generated UHDM:
```
  |uhdmallModules:
  \_module: work@unused (work@unused) ../tests/UnusedModule/top.sv:15: , parent:work@top
    |vpiDefName:work@unused
    |vpiFullName:work@unused
    |vpiPort:
    \_port: (a), line:16, parent:work@unused
      |vpiName:a
      |vpiDirection:1
      |vpiLowConn:
      \_ref_obj: 
        |vpiActual:
        \_logic_net: (@@BAD_SYMBOL@@), line:16, parent:work@unused
          |vpiName:a
          |vpiFullName:@@BAD_SYMBOL@@
          |vpiNetType:36
    |vpiContAssign:
    \_cont_assign: , line:18, parent:work@unused
      |vpiRhs:
      \_var_select: (@@BAD_SYMBOL@@), line:18
        |vpiName:a
        |vpiFullName:@@BAD_SYMBOL@@
        |vpiIndex:
        \_constant: , line:18, parent:@@BAD_SYMBOL@@
          |vpiConstType:7
          |vpiDecompile:0
          |vpiSize:32
          |INT:0
        |vpiIndex:
        \_part_select: , parent:@@BAD_SYMBOL@@
          |vpiConstantSelect:1
          |vpiParent:
          \_var_select: (@@BAD_SYMBOL@@), line:18
          |vpiLeftRange:
          \_constant: , line:18
            |vpiConstType:7
            |vpiDecompile:7
            |vpiSize:32
            |INT:7
          |vpiRightRange:
          \_constant: , line:18
            |vpiConstType:7
            |vpiDecompile:0
            |vpiSize:32
            |INT:0
      |vpiLhs:
      \_ref_obj: (@@BAD_SYMBOL@@), line:18
        |vpiName:b
        |vpiFullName:@@BAD_SYMBOL@@
    |vpiNet:
    \_logic_net: (@@BAD_SYMBOL@@), line:16, parent:work@unused
```

kamilrakoczy, 2020-11-10:
I fixed this in yosys frontend, by removing defined, but not used modules, so it is not needed for ibex project. I will leave this issue open as I think, there should be information about port width even for not used modules (if it is not the case, feel free to close this issue).

alaindargelas, 2020-11-11:
I think you did the right fix.
Surelog acts per Standard, the module "unused" is referred to in the generate statement, so it is not a top level module,
yet it does not appear in the elaboration as it is in the dead branch of the generate statement.
What you are suggesting is that more computation be done for the design parts that are in the dead branches of the elaboration tree, that is not standard and will surely yield strange results.

The flat module list uhdmallModules does not guaranty any type binding or size of objects to be correct. Only the elaborated uhdmTopModules list does so. 

I'm closing this one.

closed 2020-11-11 by alaindargelas