chipsalliance/Surelog #893: Parameters are missing for modules defined outside files parsed by Surelog.

kamilrakoczy, 2020-11-10:
There is lack of information about parameters for modules, that are defined outside files parsed by Surelog.
Test: https://github.com/antmicro/uhdm-integration/blob/missing-parameters/tests/ParameterNotExistingModule/top.sv

UHDM:
```
   |vpiModule:                                                                                                                                                                                                     
   \_module: work@top::PLLE2_ADV (work@top.pll) ../tests/ParameterNotExistingModule/top.sv:2: , parent:work@top
     |vpiDefName:work@top::PLLE2_ADV                                                                     
     |vpiName:pll                                   
     |vpiFullName:work@top.pll                                                                           
     |vpiInstance:            
     \_module: work@top (work@top) ../tests/ParameterNotExistingModule/top.sv:1: 
   |vpiArrayNet:                                                                                         
   \_array_net: (@@BAD_SYMBOL@@), line:1, parent:work@top
```

When module is defined in files parsed by Surelog, there is information about parameter (https://github.com/antmicro/uhdm-integration/blob/missing-parameters/tests/ParameterNotExistingModule/top_defined.sv):
```
   |vpiModule:                                                                                           
   \_module: work@PLLE2_ADV (work@top.pll) ../tests/ParameterNotExistingModule/top_defined.sv:2: , parent:work@top
     |vpiDefName:work@PLLE2_ADV                                                                          
     |vpiName:pll                                                                                                                                                                                                  
     |vpiFullName:work@top.pll                                                                           
     |vpiInstance:                                                                                       
     \_module: work@top (work@top) ../tests/ParameterNotExistingModule/top_defined.sv:1:                                                                                                                           
     |vpiParamAssign:                                                                                                                                                                                              
     \_param_assign: , line:7, parent:work@PLLE2_ADV                                                     
     |vpiParameter:                                                                                                                                                                                                
     \_parameter: (@@BAD_SYMBOL@@), line:3, parent:work@top.pll                                                                                                                                                    
       |vpiName:TEST                                                                                     
       |vpiFullName:@@BAD_SYMBOL@@                                                                       
       |STRING:"TEST PARAMETER"
```

alaindargelas, 2020-11-11:
Before I proceed with a hack which is to create a grey box for these modules with some information like parameters based on the usage of the undefined module (which I don't like),
let's try to actually parse the techmap library by passing it to Surelog with the -y <dir> or -v <file> Standard command line options.
Please report on the experiment. The options might need some fixing in the process.

alaindargelas, 2020-11-11:
See Surelog/tests/OldLibrary/OldLibrary.sl

mithro, 2020-11-11:
@alaindargelas - How does Surelog handle if the PLLE2_ADV is a blackbox module?

alaindargelas, 2020-11-12:
If the model is a real blackbox, like only a name is provided, then Surelog will not infer parameters or ports from its usage.
If the model is a grey box, where ports and parameters are specified, then Surelog will match the usage with the model specification.
What I see in the library is that this PLLE2_ADV is actually a full fledged model, so there should be no problem for Surelog to match it with its instantiation (modulo bugs).

alaindargelas, 2020-11-12:
@kamilrakoczy have you tried to pass the library along with the design?

kamilrakoczy, 2020-11-12:
@alaindargelas I tried to do that, but Surelog fails to parse ``(...)`` in module definition of [cells_xtra.v](https://raw.githubusercontent.com/antmicro/yosys/master/techlibs/xilinx/cells_xtra.v) file (where ``PLLE2_ADV`` is defined).
Removing ``...`` from the definitions makes Surelog to parse ``cells_xtra.v`` correctly and there are parameters in the UHDM:
```
      |vpiParameter:
      \_parameter: (@@BAD_SYMBOL@@), line:25, parent:work@clkgen_xil7series.pll
        |vpiName:BANDWIDTH
        |vpiFullName:@@BAD_SYMBOL@@
        |STRING:"OPTIMIZED"
      |vpiParameter:
      \_parameter: (@@BAD_SYMBOL@@), line:29, parent:work@clkgen_xil7series.pll
        |vpiName:CLKFBOUT_MULT
        |vpiFullName:@@BAD_SYMBOL@@
        |INT:12
...
```
Then files are correctly parsed by yosys/vivado, but bitstream is not working on HW (not sure why, need to investigate it further).

alaindargelas, 2020-11-13:
BTW, module m (...); is not part of the SystemVerilog language, Surelog is right to fail.

alaindargelas, 2020-11-16:
@kamilrakoczy, can we close this issue?

kamilrakoczy, 2020-11-16:
Yes, parameters are present when ``PLLE2_ADV`` is also parsed by Surelog, so this resolves this issue, not working bitstream is another thing that I will investigate. Closing issue.

tgorochowik, 2021-07-06:
@alaindargelas this is the issue that we discussed during the meeting. Could we revisit that and add support for this in Surelog?

alaindargelas, 2021-07-07:
@tgorochowik what do you want to do for this?
Options: 
1) Parse cells_xtra.v (support this syntax: module foo ( ... ); ) and get the parameter defintions from there?
2) Do not parse cells_xtra.v and only model the actual parameter overides present when the module is instantiated?

alaindargelas, 2021-07-07:
@tgorochowik what do you want to do for this?Options: 1) Parse cells_xtra.v (support this syntax: module foo ( ... ); ) and get the parameter defintions from there?2) Do not parse cells_xtra.v and only model the actual parameter overides present when the module is instantiated?

 
 

Reopened #893.

tgorochowik, 2021-07-07:
@alaindargelas both would work!

what we originally did was we patched `(...)` to `()` which would be equivalent to point 1.

However if we went with 2. it would probably allow for greater flexibility - for example it should be easier to mix different frontends (sv and uhdm) - but we would need to store the information in uhdm to know whether a given module was fully parsed by Surelog or not.

So to answer your questions, there are some benefits to 2. but both options should be okay

alaindargelas, 2021-07-08:
Fixed by #1409 with option 2:
 Do not parse cells_xtra.v and only model the actual parameter overides present when the module is instantiated

closed 2021-07-08 by alaindargelas