chipsalliance/Surelog #894: struct type not recognized correctly

mjc0608, 2020-11-11:
Using the following command line
``` bash
surelog -parse -sverilog test.sv
```
to parse the following file
``` verilog
package test_pkg;

typedef struct packed {
    logic valid;
    logic [63:0] data;
} t_test;
parameter TEST_SIZE = $bits(t_test);

endpackage

```
leads to the following error
``` bash
test.sv:7: Undefined variable: t_test.
```
I think this file can be correctly parsed by VCS and verilator. It seems to be a bug in Surelog.

alaindargelas, 2020-11-11:
I'll fix the error message (which is bogus).

But please note that the UHDM model is correct (TEST_SIZE = 65):

surelog -parse -sverilog test.sv -d uhdm  -d vpi_ids -elabuhdm

```
====== UHDM =======
design: (unnamed), id:11
|vpiName:unnamed
|uhdmallPackages:
\_package: test_pkg (test_pkg), id:12 tests/ScratchPad.sv:1: , parent:unnamed, parID:11
  |vpiDefName:test_pkg
  |vpiFullName:test_pkg
  |vpiTypedef:
  \_struct_typespec: (t_test), id:0, line:3
    |vpiPacked:1
    |vpiName:t_test
    |vpiTypespecMember:
    \_typespec_member: (valid), id:2, line:4
      |vpiName:valid
      |vpiTypespec:
      \_logic_typespec: , id:1, line:4
    |vpiTypespecMember:
    \_typespec_member: (data), id:7, line:5
      |vpiName:data
      |vpiTypespec:
      \_logic_typespec: , id:6, line:5
        |vpiRange:
        \_range: , id:3, line:5, parent:t_test, parID:0
          |vpiLeftRange:
          \_constant: , id:4, line:5, parID:3
            |vpiConstType:7
            |vpiDecompile:63
            |vpiSize:64
            |INT:63
          |vpiRightRange:
          \_constant: , id:5, line:5, parID:3
            |vpiConstType:7
            |vpiDecompile:0
            |vpiSize:64
            |INT:0
  |vpiParamAssign:
  \_param_assign: , id:9, line:7, parent:test_pkg, parID:12
    |vpiRhs:
    \_constant: , id:10, line:7
      |vpiDecompile:65
      |INT:65
    |vpiLhs:
    \_parameter: (test_pkg::TEST_SIZE), id:8, line:7, parent:test_pkg, parID:12
      |vpiName:TEST_SIZE
      |vpiFullName:test_pkg::TEST_SIZE
  |vpiParameter:
  \_parameter: (test_pkg::TEST_SIZE), id:8, line:7, parent:test_pkg, parID:12
```

closed 2020-11-13 by alaindargelas

mjc0608, 2020-11-13:
Still have some similar issues. I will upload a more concrete example.

mjc0608, 2020-11-13:
If you untar the attached file, enter the directory, and use `surelog -parse -sverilog -elabuhdm -f sources.txt` to parse these files. Surelog will print a bunch of error with regards to "undefined type". However, the generated `.uhdm` files seem correct.

Sorry that this is the only example I have in my hand. It's a little bit heavy.
[demo.tar.gz](https://github.com/alainmarcel/Surelog/files/5539130/demo.tar.gz)

PR #896 (2020-11-13): Fix bogus error msg in evaluation
Fixes #894