chipsalliance/Surelog #897: Github actions have been disabled by GitHub staff for Surelog

alaindargelas, 2020-11-13:
@umarcor, can you check and remedy?
Surelog has no more ci check.

umarcor, 2020-11-13:
Shall I contact GitHub on behalf of this project?

alaindargelas, 2020-11-13:
Yes, please do.

alaindargelas, 2020-11-13:
Are we using too much compute resources?

umarcor, 2020-11-13:
Honestly, no idea. This is the first project I've seen disabled by GitHub staff. I've only seen Actions being temporally stalled in other projects, because the 20 parallel job limit was reached. Let's see what reply we get.

alaindargelas, 2020-11-14:
I think it's because I prematurely committed before the task were finished.
Github terminated the jobs. My bad!

closed 2020-11-14 by alaindargelas

umarcor, 2020-11-14:
@alaindargelas it was an unfortunate error on their side. They replied to me last night and they used sentences such as "*sorry to hear you've experienced this issue*", "*internal issue that has since been fixed*", "*if you still encounter this error message tomorrow, please reach out and let us know*", "*we're terribly sorry for this*".

So, nothing it worry about. There is no problem with commiting prematurely, or with the time/resource limits. Just some human interactions that are required from time to time 😄