chipsalliance/Surelog #899: logic typedef becomes wire

mjc0608, 2020-11-14:
While using `surelog --Mdir tttt -parse -elabuhdm -sverilog test.sv` to parse the following file:
``` verilog
module test;
    typedef logic [1:0] t_two_bits;
    t_two_bits kkkk;
endmodule
```
Surelog generates the following uhdm:
```
...
|uhdmtopModules:
\_module: work@test (work@test) test.sv:1: 
  |vpiDefName:work@test
  |vpiName:work@test
  |vpiVariables:
  \_logic_var: (@@BAD_SYMBOL@@), line:13, parent:work@test
    |vpiName:kkkk
    |vpiFullName:@@BAD_SYMBOL@@
    |vpiAutomatic:1
    |vpiVisibility:1
  |vpiTypedef:
  \_logic_typespec: (t_two_bits), line:2, parent:work@test
    |vpiName:t_two_bits
    |vpiRange:
    \_range: , line:2, parent:t_two_bits
      |vpiLeftRange:
      \_constant: , line:2
        |vpiConstType:7
        |vpiDecompile:1
        |vpiSize:64
        |INT:1
      |vpiRightRange:
      \_constant: , line:2
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:64
        |INT:0
```
It seems like the variable `kkkk` with type `t_two_bits` is instantiated with type `wire`.

mjc0608, 2020-11-14:
Also, the generated uhdm file cannot be parsed by yosys, because vpiLogicTypespec is not handled correctly. I will submit a pull request for that part.

alaindargelas, 2020-11-14:
1) @mjc0608 may I suggest you make a full build: rm -rf build dbuild; make; make debug
You have BAD_SYMBOL in your trace, but if I run your example I have the correct name.

2) You mean it has a type logic_var and you expect type logic (like a logic net)?

mjc0608, 2020-11-14:
1. I did a rebuild and the BAD_SYMBOL is still there. What's the command line you use?

2. If you read the file via yosys directly, yosys gives the following AST.
```
    AST_MODULE <a.v:1.1-4.10> [0x5609ee3ca470] str='\test'
      AST_TYPEDEF <a.v:0.0-0.0> [0x5609ee3c8890] str='\t_two_bits'
        AST_WIRE <a.v:0.0-0.0> [0x5609ee3c8e70] logic
          AST_RANGE <a.v:0.0-0.0> [0x5609ee3c89d0]
            AST_CONSTANT <a.v:2.17-2.18> [0x5609ee3c8730] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1
            AST_CONSTANT <a.v:2.19-2.20> [0x5609ee3c8d30] bits='00000000000000000000000000000000'(32) signed range=[31:0]
      AST_WIRE <a.v:3.13-3.17> [0x5609ee29fa70] str='\kkkk'
        AST_WIRETYPE <a.v:0.0-0.0> [0x5609ee3c9970] str='\t_two_bits'
```
which means the type of `kkkk` is defined as `t_two_bits`. If this information is lost in Surelog, it would be impossible to reconstruct in yosys.

mjc0608, 2020-11-14:
Also, it would be helpful if you can provide the UHDM commit you are using. It could be an issue on the UHDM side.

alaindargelas, 2020-11-15:
#902 should fix your problem,
Take a look at Surelog/tests/LogicTypedef/LogicTypedef.log

After pulling and recompiling, run that test yourself:
cd Surelog/build
../tests/regression.tcl test=LogicTypedef

Check your local output:
Surelog/build/tests/LogicTypedef/LogicTypedef.log for BAD_SYMBOL, if you still have them, then something really bad is happening with your build.

I have the latest UHDM and Surelog checked out.

alaindargelas, 2020-11-15:
Use this command: 
cd Surelog/tests/LogicTypedef/
../../build/bin/surelog --Mdir tttt -parse -elabuhdm -sverilog dut.sv -nobuiltin -d uhdm

alaindargelas, 2020-11-15:
Apologies, I now see the BAD_SYMBOLS when using uhdm-dump instead of the built-in surelog -d uhdm

alaindargelas, 2020-11-16:
This is now fixed along with the BAD_SYMBOLS

closed 2020-11-16 by alaindargelas