chipsalliance/Surelog #922: Wrong order in multilevel ternary operator

kamilrakoczy, 2020-11-24:
When using multilevel ternary operator (``cond ? res1 : cond2 ? res2 : res3``) Surelog generates UHDM in wrong order.

Test: https://github.com/antmicro/uhdm-integration/blob/ternary/tests/TernaryOrder/top.sv
UHDM:
```
   |vpiContAssign:                                                                                       
     |vpiRhs:                
       |vpiOpType:vpiConditionOp
       |vpiOperand:
         |vpiOpType:vpiConditionOp
         |vpiOperand:
         \_ref_obj: (work@top.first_condition), line:4
           |vpiName:first_condition
           |vpiFullName:work@top.first_condition
         |vpiOperand:
         \_ref_obj: (work@top.first_condition_true), line:4
           |vpiName:first_condition_true
           |vpiFullName:work@top.first_condition_true
         |vpiOperand:
         \_ref_obj: (work@top.first_condition_false), line:5
           |vpiName:first_condition_false
           |vpiFullName:work@top.first_condition_false
       |vpiOperand:
       \_ref_obj: (work@top.second_condition_true), line:5
         |vpiName:second_condition_true
         |vpiFullName:work@top.second_condition_true                                                                                                                                                                      
       |vpiOperand:
       \_ref_obj: (work@top.second_condition_false), line:6
         |vpiName:second_condition_false
         |vpiFullName:work@top.second_condition_false
     |vpiLhs:
     \_ref_obj: (work@top.result), line:4
       |vpiName:result
       |vpiFullName:work@top.result
```
this uhdm corresponds to: 
```
result = (first_condition ? first_condition_true : first_condition_false) ? second_condition_true : second_condition_false
```
but the test corresponds to:
```
result = first_condition ? first_condition_true : (first_condition_false ? second_condition_true : second_condition_false)
```

Thomasb81, 2020-11-24:
I think ternary and power operator are right associative

so
https://github.com/alainmarcel/Surelog/blob/3cbb369a2d5929208d91a1100c2b3b79fce35b39/grammar/SV3_1aParser.g4#L2887

should be : 
`    | <assoc=right> constant_expression ( LOGICAL_AND expression )* conditional_operator ( attribute_instance )* expression COLUMN constant_expression `

Probably this one also:
https://github.com/alainmarcel/Surelog/blob/3cbb369a2d5929208d91a1100c2b3b79fce35b39/grammar/SV3_1aParser.g4#L3004
But I have not in mind the context of this rule.

This impact also the '**' operator as far as I know.

This way the antlr AST is expected to be build in the right order.

alaindargelas, 2020-11-25:
Fixed by #923

closed 2020-11-25 by alaindargelas