chipsalliance/Surelog #926: PP make bad space removing for number

DamienLeBars, 2020-11-26:
In "##2 1'b1" the space must be kept, but Surelog preprocessing do:
Before preprocessing:
```verilog
module top;
property inq2(r1,r2);
@(posedge clk) a ##[r1:r2] b ##2 1'b1 |=> d;
endproperty
endmodule
```
Afterpreprocessing:
```verilog
module top;
property inq2(r1,r2);
@(posedge clk) a ##[r1:r2] b **##21'b1** |=> d;
endproperty
endmodule
```
`PP TREE part:
(description (text_blob ##)) (description (number 2 1)) (description (number 'b1))`

I don't know why number can have spaces in the lexer def:
[lexer PP number](https://github.com/alainmarcel/Surelog/blob/4dc19edbf640e7fe7ce7cf8545b1a1f77b54ab3b/grammar/SV3_1aPpLexer.g4#L183)

If the lexer is ok, the problem could be more from "enter number listener" function:
[SV3_1aPpTreeShapeListener.cpp](https://github.com/alainmarcel/Surelog/blob/4dc19edbf640e7fe7ce7cf8545b1a1f77b54ab3b/src/SourceCompile/SV3_1aPpTreeShapeListener.cpp#L99)

alaindargelas, 2020-11-27:
"I don't know why number can have spaces in the lexer def" - That is a nasty feature of Verilog: you can have spaces in a number, that's why the lexer has to support it.
We probably have to introduce a new Lexer Token before '##' and the integer tokens:   '##[0-9]+', then amend the corresponding Parser rules to accept it and do a bit of postprocessing in the listener to get the value...

Thomasb81, 2020-11-27:
> That is a nasty feature of Verilog: you can have spaces in a number, that's why the lexer has to support it.

Can you provide a pointer to the standard on this ?  I am surprise and can't find anything on this. While number can contain underscore.

alaindargelas, 2020-11-27:
module toto();
  wire a;
  assign a = 2'b 10;
   assign a = 2 'b 10 ;
  assign a = 2 'b _10 ;
endmodule

You can try the above in any simulator. Granted, spaces are not allowed in between the digits. I think the Icarus regression contains the most nasty examples.

Thomasb81, 2020-11-27:
The way I understand the BNF SV1800_2017, At least `2 'b _10` is illegal. 

Rule binary_value / page 72:
`binary_value := binary_digit { _ | binary_digit }`

Note in HdlConvertor we have address the preprocessor problem with island/mode feature of antlr. 
https://github.com/Nic30/hdlConvertor/blob/master/grammars/verilogPreprocLexer.g4

This allow depending of the context to consider or not some token, that interest the preprocessor grammar or not.

alaindargelas, 2020-11-27:
This number example works in edaplaygound with any simulator, so it's legal, the bnf is sometimes too formal and the actual commercial tools went above and beyond. There are more nasty examples in Surelog/third_party/tests/Icarus/* that are legal. Icarus collected those and compared with commercial simulators.

I checked the island feature before, does it solve the problem on this case? Did you try the test summited in this issue?

Thomasb81, 2020-11-27:
I know a lot of syntax initially supported by verilog-xl but not reported in the bnf are de facto "standard".

I think hldConvertor pass the icarus testsuite ... At least it was used to test verilog preprocessor and grammar support.

> I checked the island feature before, does it solve the problem on this case? Did you try the test summited in this issue?
I expect so : outside of preprocessor substitution space are expected to be preserved. But definitively I trial should confirm this.

PR #930 (2020-11-29): ##delay support
Fixes #926
closed 2020-11-29 by alaindargelas