chipsalliance/Surelog #927: PP: block comment over `else generate error

DamienLeBars, 2020-11-26:
Before preprocessing:
```verilog
   //`define USE_BOOTMODE_TEST
   `ifdef USE_BOOTMODE_TEST
      assign a= b;
      assign c= d;/*
   `else
      assign a= 1'b0;
      assign c= 1'b0;*/
   `endif
```
Afterpreprocessing:
```verilog
      assign a= 1'b0;
      assign c= 1'b0;*/
```
The "*/" generates a parsing error.
PP should probably be disabled inside comments (I think my case is not the worst because it generates an error, but we can easily imagine worst cases that are dangerously noiseless).

alaindargelas, 2020-11-27:
The PP grammar failed completely on that one, it would take some major re-haul to support this properly.
Unless you find some real example where it silently creates a bad output that the parser does not error out on, I suggest to leave it be.
You can experiment with trying to support it better by introducing new rules that detect the pattern:

cd Surelog/grammar
"Edit SV3_1aLexer.g4"
@lexer::members {
// C++ target:
//   bool sverilog;
// Java target:
Boolean sverilog = true;
}

ant compile_java
ant javac
ant test_pp_tree
Buildfile: /home/alain/Surelog/grammar/build.xml

test_pp_tree:

"Cut and paste"

//`define USE_BOOTMODE_TEST
   `ifdef USE_BOOTMODE_TEST
      assign a= b;
      assign c= d;/*
   `else
      assign a= 1'b0;
      assign c= 1'b0;*/
   `endif

"Hit Ctrl-D"

     [java] (source_text (description (comments //`define USE_BOOTMODE_TEST\n)) (description (text_blob    )) (description (ifdef_directive `ifdef   USE_BOOTMODE_TEST)) (description (text_blob \n)) (description (text_blob       )) (description (text_blob assign)) (description (text_blob  )) (description (text_blob a)) (description (text_blob =)) (description (text_blob  )) (description (text_blob b)) (description (text_blob ;)) (description (text_blob \n)) (description (text_blob       )) (description (text_blob assign)) (description (text_blob  )) (description (text_blob c)) (description (text_blob =)) (description (text_blob  )) (description (text_blob d)) (description (text_blob ;/*)) (description (text_blob \n)) (description (text_blob    )) (description (else_directive `else)) (description (text_blob \n)) (description (text_blob       )) (description (text_blob assign)) (description (text_blob  )) (description (text_blob a)) (description (text_blob =)) (description (text_blob  )) (description (number 1'b0)) (description (text_blob ;)) (description (text_blob \n)) (description (text_blob       )) (description (text_blob assign)) (description (text_blob  )) (description (text_blob c)) (description (text_blob =)) (description (text_blob  )) (description (number 1'b0)) (description (text_blob ;*/)) (description (text_blob \n)) (description (text_blob    )) (description (endif_directive `endif)) (description (text_blob \n)))

alaindargelas, 2020-11-27:
The PP didn't recognized any structure as you can see from the rule dump.

PR #931 (2020-11-30): Fix PP issue with comments
Fixes #927.
That was not so complicated a change. The PP was to greedy in consuming all the characters in the sequence: ;/*
closed 2020-11-30 by alaindargelas