chipsalliance/Surelog #932: ModPortRange regression

DamienLeBars, 2020-11-30:
Hi,
i have detected a regression here:
[History for Surelog/tests/ModPortRange/ModPortRange.log](https://github.com/alainmarcel/Surelog/commits/master/tests/ModPortRange/ModPortRange.log)
The [Remove bogus nets in elaborated model](https://github.com/alainmarcel/Surelog/commit/a188d0064f62c87dd70055abb465ce0484c9e1ed#diff-c4bcbbbc1cd874663899bac800d61f8e44a8f452703f1105717af4160a159fae) removed access to the modport range that was fixed by [Mod port range bug](https://github.com/alainmarcel/Surelog/commit/97ce0e1a94fa321bd8c7afe6e1a7eda0470dfdb8#diff-c4bcbbbc1cd874663899bac800d61f8e44a8f452703f1105717af4160a159fae) .

alaindargelas, 2020-12-01:
The previously reported array_net with range is not compliant with the standard. The fix I made that regressed this change was also not complete.
To fix this correctly and the other problem of extra nets created by modport is going to be quite involved.
I have to start using "interface typespec" in the definition of ports that involve interface or modports and create the proper lowconn to hold the dimensions as ports don't have dimensions in the Object Model.

DamienLeBars, 2020-12-01:
Ok, i understand that even if it bothers me a bit, the standard is not easy to read on this subject.

alaindargelas, 2020-12-04:
#941 should fix this, please confirm.
The model uses interface_array, interface and modport objects to hold all the information.
You can check the log of ModPortRange as a complete testcase.

alaindargelas, 2020-12-05:
@DamienLeBars, I'll wait your confirmation before closing.

DamienLeBars, 2020-12-14:
> 
> 
> #941 should fix this, please confirm.
> The model uses interface_array, interface and modport objects to hold all the information.
> You can check the log of ModPortRange as a complete testcase.

Thanks, it looks fine to me.

closed 2020-12-14 by DamienLeBars