chipsalliance/Surelog #937: Missing way of setting parameters when parsing file.

kamilrakoczy, 2020-12-02:
I couldn't find a way to set/change parameters when parsing file using Surelog.

It is necessary, for Ibex to set path to ``SRAMInitFile``. Without this, Surelog removes all parts related with setting mem file, as ``SRAMInitFile`` parameter is not set.

alaindargelas, 2020-12-03:
-Dvar=value

kamilrakoczy, 2020-12-03:
It seems doesn't work (or am I doing it wrong?).

Run cmd:
```
surelog -parse -sverilog \
     -v cells_xtra_surelog.v \
     -DSRAMInitFile="/home/grey/uhdm/uhdm-integration/led.vmem" \
     clkgen_xil7series.sv ...
```

and then checking every entry of ``SRAMInitFile`` (or ``MemInitFile`` that should be set to ``SRAMInitFile`` value) is empty:
```
581457    \_parameter: (work@ram_1p.MemInitFile), line:13, parent:work@ram_1p
581458    |vpiParameter:

582527    |vpiParamAssign:
582528    \_param_assign: , line:14, parent:work@top_artya7
582529      |vpiRhs:
582530      \_constant: , line:14
582531        |vpiConstType:6
582532        |STRING:
582533      |vpiLhs:
582534      \_parameter: (work@top_artya7.SRAMInitFile), line:14, parent:work@top_artya7
582535        |vpiName:SRAMInitFile
582536        |vpiFullName:work@top_artya7.SRAMInitFile

613597    \_parameter: (work@top_artya7.SRAMInitFile), line:14, parent:work@top_artya7
613598      |vpiName:SRAMInitFile
613599      |vpiFullName:work@top_artya7.SRAMInitFile
613600      |STRING:

612690      |vpiParameter:
612691      \_parameter: (work@top_artya7.u_ram.MemInitFile), line:111, parent:work@top_artya7.u_ram
612692        |vpiName:MemInitFile
612693        |vpiFullName:work@top_artya7.u_ram.MemInitFile
612694        |STRING:

612621        |vpiParameter:
612622        \_parameter: (work@top_artya7.u_ram.u_ram.MemInitFile), line:58, parent:work@top_artya7.u_ram.u_ram
612623          |vpiName:MemInitFile
612624          |vpiFullName:work@top_artya7.u_ram.u_ram.MemInitFile
612625          |STRING:

612563              |vpiParameter:
612564              \_parameter: (work@top_artya7.u_ram.u_ram.gen_generic.u_impl_generic.MemInitFile), line:46, parent:work@top_artya7.u_ram.u_ram.gen_generic.u_impl_generic
612565                |vpiName:MemInitFile
612566                |vpiFullName:work@top_artya7.u_ram.u_ram.gen_generic.u_impl_generic.MemInitFile
612567                |STRING:
```
and then looking at generated AST, this parameters are also empty:
```
AST_PARAMETER  str='\SRAMInitFile' basic_prep
   AST_CONSTANT basic_prep range=[-1:0]
...
AST_PARAMETER str='\MemInitFile' basic_prep
   AST_CONSTANT basic_prep range=[-1:0]
```

alaindargelas, 2020-12-03:
I see, this is totally not supported at this time.
Also, I'll need a pointer to an actual testcase on github with all these files:

cells_xtra_surelog.v 
-DSRAMInitFile="/home/grey/uhdm/uhdm-integration/led.vmem" 
clkgen_xil7series.sv

kamilrakoczy, 2020-12-03:
I created extra branch test in uhdm-integration: https://github.com/antmicro/uhdm-integration/tree/set-parameter-test
In order to run test, you should type:
```
make TEST=tests/ibex uhdm/yosys/synth-ibex
```
This will build necessary binaries, configure ibex, run Surelog and then run yosys on generated uhdm file.
Makefile that is configuring Surelog is available: https://github.com/antmicro/uhdm-integration/blob/set-parameter-test/tests/ibex/Makefile.in#L37

alaindargelas, 2020-12-04:
In the interest of (my) time, I don't need to do all of this, I simply need the path to the 3 files:
cells_xtra_surelog.v
led.vmem
clkgen_xil7series.sv

Can you attach these files here or check them in so I can retrieve them simply by cloning the repo and without having to build the whole thing?

kamilrakoczy, 2020-12-04:
Sorry that I made you confusion. Run cmd that I posted above is shorted and it is actually whole Ibex synthesis. 

I prepared simple test showing the problem:
```
module dut (output logic o);
   parameter SRAMInitFile = "";
   ram_1p #(
     .MemInitFile(SRAMInitFile)
   ) u_ram (
           .out(o)
   );
endmodule

module ram_1p #(
        parameter     MemInitFile = ""
) (
        output logic out
);

initial begin
  if (MemInitFile != "") begin : gen_meminit
      $display("Initializing memory %m from file '%s'.", MemInitFile);
      assign out = 1'b1;
  end

  if (MemInitFile == "") begin : gen_nomeminit
      $display("Initializing memory file is empty.");
      assign out = 1'b0;
  end
end

endmodule
```

I also packed files directly from Ibex that are using ``SRAMInitFile`` and ``MemInitFile`` (but I believe they are not needed anymore with the test above) : [parameters.tar.gz](https://github.com/alainmarcel/Surelog/files/5641176/parameters.tar.gz)

alaindargelas, 2020-12-05:
Verilog does not allow passing parameter from command line directly.
You need a macro:

```
`define PARAM_FILE ""

module dut (output logic o);
   parameter SRAMInitFile = `PARAM_FILE;
   ram_1p #(
     .MemInitFile(SRAMInitFile)
   ) u_ram (
           .out(o)
   );
endmodule

module ram_1p #(
        parameter     MemInitFile = ""
) (
        output logic out
);

initial begin
  if (MemInitFile != "") begin : gen_meminit
      $display("Initializing memory %m from file '%s'.", MemInitFile);
      assign out = 1'b1;
  end

  if (MemInitFile == "") begin : gen_nomeminit
      $display("Initializing memory file is empty.");
      assign out = 1'b0;
  end
end

endmodule
```
Run:
./dbuild/bin/surelog tests/ScratchPad.sv -parse -d uhdm -nocache -DPARAM_FILE=\"BLAH.v\"

You get the correct result:

```
|uhdmallModules:
\_module: work@dut (work@dut) tests/ScratchPad.sv:5: , parent:work@dut
  |vpiDefName:work@dut
  |vpiFullName:work@dut
  |vpiPort:
  \_port: (o), line:5, parent:work@dut
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@dut.o), line:5, parent:work@dut
        |vpiName:o
        |vpiFullName:work@dut.o
        |vpiNetType:36
  |vpiNet:
  \_logic_net: (work@dut.o), line:5, parent:work@dut
  |vpiParamAssign:
  \_param_assign: , line:6, parent:work@dut
    |vpiRhs:
    \_constant: , line:6
      |vpiConstType:6
      |vpiDecompile:BLAH.v
      |vpiSize:6
      |STRING:BLAH.v
    |vpiLhs:
    \_parameter: (work@dut.SRAMInitFile), line:6, parent:work@dut
      |vpiName:SRAMInitFile
      |vpiFullName:work@dut.SRAMInitFile
[... 168 lines cut ...]
    \_parameter: (work@dut.u_ram.MemInitFile), line:8, parent:work@dut.u_ram
      |vpiName:MemInitFile
      |vpiFullName:work@dut.u_ram.MemInitFile
      |STRING:BLAH.v
  |vpiNet:
  \_logic_net: (work@dut.o), line:5, parent:work@dut
  |vpiParamAssign:
  \_param_assign: , line:6, parent:work@dut
    |vpiRhs:
    \_constant: , line:6
      |vpiConstType:7
      |vpiDecompile:BLAH.v
      |vpiSize:8
      |STRING:BLAH.v
    |vpiLhs:
    \_parameter: (work@dut.SRAMInitFile), line:6, parent:work@dut
      |vpiName:SRAMInitFile
      |vpiFullName:work@dut.SRAMInitFile
  |vpiParameter:
  \_parameter: (work@dut.SRAMInitFile), line:6, parent:work@dut
    |vpiName:SRAMInitFile
    |vpiFullName:work@dut.SRAMInitFile
    |STRING:BLAH.v

```

kgugala, 2020-12-08:
@alaindargelas both Yosys and Verilator support parameters overwriting from command line (also some proprietary tools).

A lot of projects actually use this functionality. I think we should consider handling the functionality of overwriting params with command line switches. Otherwise, we'd have to ask people to modify their designs to use it with Surelog/UHDM (I believe this is does not sound like something we'd like to do)

BTW this is the last issue blocking us from synthesising (unmodified) Ibex with Surelog/UHDM -> Yosys flow.

(CC @mithro @hzeller)

alaindargelas, 2020-12-08:
Alright, I'll use the -Dparam=value command.

tgorochowik, 2020-12-08:
See verilator for reference:
* https://github.com/verilator/verilator/blob/master/bin/verilator#L1363-L1371
* https://github.com/verilator/verilator/blob/master/bin/verilator#L340

PR #953 (2020-12-09): -Pparam=value and -top options
Fixes #937
closed 2020-12-09 by alaindargelas