chipsalliance/Surelog #943: Incorrect constant size in assignment

rkapuscik, 2020-12-04:
Consider this line: https://github.com/lowRISC/ibex/blob/master/rtl/ibex_counter.sv#L61
```verilog
counter_q <= '0;
```
`counter_q` is defined as `logic [CounterWidth-1:0] counter_q;`, `CounterWidth` is 32.
The UHDM fragment for the assignment is:
```
         |vpiStmt:
          \_begin: (work@ibex_counter), line:60
            |vpiFullName:work@ibex_counter
            |vpiStmt:
            \_assignment: , line:61, parent:work@ibex_counter
              |vpiOpType:82
              |vpiLhs:
              \_ref_obj: (work@ibex_counter.counter_q), line:61, parent:work@ibex_counter
                |vpiName:counter_q
                |vpiFullName:work@ibex_counter.counter_q
                |vpiActual:
                \_logic_net: (work@ibex_counter.counter_q), line:53, parent:work@ibex_counter
                  |vpiName:counter_q
                  |vpiFullName:work@ibex_counter.counter_q
                  |vpiNetType:36
                  |vpiRange:
                  \_range: , line:53
                    |vpiLeftRange:
                    \_constant: , line:53
                      |vpiConstType:7
                      |vpiDecompile:31
                      |vpiSize:64
                      |INT:31
                    |vpiRightRange:
                    \_constant: , line:53
                      |vpiConstType:7
                      |vpiDecompile:0
                      |vpiSize:64
                      |INT:0
              |vpiRhs:
              \_constant: , line:61
                |vpiConstType:3
                |vpiDecompile:'b0
                |vpiSize:1
                |BIN:0
```
Constant in vpiRhs has incorrect size, it should be expanded to fit vpiLhs.

alaindargelas, 2020-12-05:
I'd rather leave this task to the client application(s), when they require expression sizing (padding), based on assignment, to be performed.
The Verilog rules are obscure and I was hopping Verilator and Yosys already performed such adjustments.
If it is absolutely required, this warrant a new type of analysis for expression computation with Verilog padding rules that is not currently in scope.

alaindargelas, 2020-12-05:
https://comp.lang.verilog.narkive.com/yh56Vezq/implicit-zero-padding

alaindargelas, 2020-12-06:
UHDM carries all the necessary row information to perform this computation.
I suggest we implement this as a decorator on UHDM like the Elaborator if i's needed.

rkapuscik, 2020-12-07:
I think it could be left to the client application, but there are some inconsistencies to handle first, for example `1'b0` and `'0` are stored the same in UHDM, as:
```
\_constant: , line:61
  |vpiConstType:3
  |vpiDecompile:'b0
  |vpiSize:1
  |BIN:0
```
Then in some situations this needs to be parsed as unsized (assignments) or sized (concatenations).
Maybe it will be enough to somehow distinguish the unsized constants before implementing the optional pass?

PR #952 (2020-12-08): set unsized values with a 0 size
Fixes #943
closed 2020-12-08 by alaindargelas