chipsalliance/Surelog #955: EnumConsts have incorrect values

rkapuscik, 2020-12-09:
In enum declared here: https://github.com/lowRISC/ibex/blob/master/rtl/ibex_pkg.sv#L278
```
typedef enum logic [5:0] {
  EXC_CAUSE_IRQ_SOFTWARE_M     = {1'b1, 5'd03},
  EXC_CAUSE_IRQ_TIMER_M        = {1'b1, 5'd07},
(...)
```
Created UHDM:
```
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_IRQ_SOFTWARE_M), line:264
      |vpiName:EXC_CAUSE_IRQ_SOFTWARE_M
      |INT:100003
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_IRQ_TIMER_M), line:265
      |vpiName:EXC_CAUSE_IRQ_TIMER_M
      |INT:100007
```
The EnumConst node value is not correct, I expected it to be stored in similar way as normal vpiConst nodes.

rkapuscik, 2020-12-10:
The values are still incorrect:
```
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_BREAKPOINT), line:273
      |vpiName:EXC_CAUSE_BREAKPOINT
      |BIN:00011
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_ECALL_MMODE), line:277
      |vpiName:EXC_CAUSE_ECALL_MMODE
      |BIN:01101
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_ECALL_UMODE), line:276
      |vpiName:EXC_CAUSE_ECALL_UMODE
      |BIN:00001
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_ILLEGAL_INSN), line:272
      |vpiName:EXC_CAUSE_ILLEGAL_INSN
      |BIN:00001
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_INSN_ADDR_MISA), line:270
      |vpiName:EXC_CAUSE_INSN_ADDR_MISA
      |BIN:00000
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_INSTR_ACCESS_FAULT), line:271
      |vpiName:EXC_CAUSE_INSTR_ACCESS_FAULT
      |BIN:00001
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_IRQ_EXTERNAL_M), line:266
      |vpiName:EXC_CAUSE_IRQ_EXTERNAL_M
      |BIN:101101
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_IRQ_NM), line:269
      |vpiName:EXC_CAUSE_IRQ_NM
      |BIN:111111
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_IRQ_SOFTWARE_M), line:264
      |vpiName:EXC_CAUSE_IRQ_SOFTWARE_M
      |BIN:100011
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_IRQ_TIMER_M), line:265
      |vpiName:EXC_CAUSE_IRQ_TIMER_M
      |BIN:100111
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_LOAD_ACCESS_FAULT), line:274
      |vpiName:EXC_CAUSE_LOAD_ACCESS_FAULT
      |BIN:00101
    |vpiEnumConst:
    \_enum_const: (EXC_CAUSE_STORE_ACCESS_FAULT), line:275
      |vpiName:EXC_CAUSE_STORE_ACCESS_FAULT
      |BIN:00111
```
Notice the overlapping values and incorrect widths - it seems that when concatenated with `1'b0` the calculation takes only part of the value.
For Verilator use case it would actually be possible to pass the concatenation nodes and let it determine the value on its own, but I don't see a way to do that in the standard.

alaindargelas, 2020-12-11:
#957 should fix this issue, @rkapuscik please confirm.

rkapuscik, 2020-12-14:
Values are correct now, thanks for the fix.

PR #956 (2020-12-10): Enum const bug
Fixes #955.
Enum const have little information, they don't have an explicit type field or size, but it can be derived from the value.
closed 2020-12-14 by rkapuscik