chipsalliance/Surelog #967: Wrong format for vpiDecStrVal

kamilrakoczy, 2020-12-16:
Test:
```
module top(input a, output b);
        bit first = 1;
        logic [28:0] second = 29'd1;
endmodule
```
UHDM:
```
  |vpiContAssign:
  \_cont_assign: , line:3, parent:work@top
    |vpiRhs:
    \_constant: , line:3
      |vpiConstType:7
      |vpiDecompile:29'd1
      |vpiSize:29
      |INT:1
    |vpiLhs:
    \_logic_net: (work@top.second), line:3, parent:work@top
      |vpiName:second
      |vpiFullName:work@top.second
      |vpiNetType:36
      |vpiRange:
      \_range: , line:3
        |vpiLeftRange:
        \_constant: , line:3
          |vpiConstType:7
          |vpiDecompile:28
          |vpiSize:64
          |INT:28
        |vpiRightRange:
        \_constant: , line:3
          |vpiConstType:7
          |vpiDecompile:0
          |vpiSize:64
          |INT:0
```
reported format is ``vpiIntVal`` but it should be ``vpiDecStrVal``.

alaindargelas, 2020-12-17:
Surelog does not perform expression padding or lhs vs rhs type checking, it does report the raw expression like seen in the source code. I expect the 2 main client applications: Yosys and Verilator have the capability to do so already.

kamilrakoczy, 2020-12-17:
Ok, I agree, in this case it should return ``vpiIntVal``, according to standard (and my understanding of it):
```
When the format field is vpiObjTypeVal, the routine shall fill in the value and change the format field
based on the object type, as follows:
— For an integer, vpiIntVal
— For a real, vpiRealVal
— For a scalar, either vpiScalar or vpiStrength
— For a time variable, vpiTimeVal with vpiSimTime
— For a vector, vpiVectorVal
```
so I believe, there is some other problem with formats returned by Surelog.
Test:
```
module top(input a, output b);
        logic [11:0] first  = 12'h1; //returned format: vpiHexStrVal
        logic [11:0] second  = 12'd1; //returned format: vpiIntVal
        logic [11:0] third  = 12'b1; //returned format: vpiBinStrVal
        logic [11:0] forth  = 12'o1; //returned format: vpiOctStrVal
endmodule
```
according to standard, I think, they all should return ``vpiIntVal`` format.

alaindargelas, 2020-12-18:
I think UHDM models this correctly as following vpiConsType:

#define vpiDecConst                1   /* decimal integer */
#define vpiRealConst               2   /* real */
#define vpiBinaryConst             3   /* binary integer */
#define vpiOctConst                4   /* octal integer */
#define vpiHexConst                5   /* hexadecimal integer */

I don't see the point in changing to vpiHexStrVal.....

alaindargelas, 2020-12-18:
Are you (Or am I) mistaking Simulation runtime values (vpiHexStrVal) with elaboration const values (vpiHexConst)?
Surelog only deals with Elaboration time values.

kamilrakoczy, 2020-12-18:
I'm talking about ``vpi_get_value() `` function (chapter 38.15, page 1065 in IEEE: http://ecee.colorado.edu/~mathys/ecen2350/IntelSoftware/pdf/IEEE_Std1800-2017_8299595.pdf#page=1065).
Description of it says: ``Retrieve the simulation value of an object.`` so I believe it is simulation runtime value.

We are using this function to get values of e.g. parameters, constants etc:
```
s_vpi_value val;
vpi_get_value(obj_h, &val);
```
``vpi_get_value`` is also setting format field ``val.format`` (and I'm talking about this field). Based on this field, we are differently parsing value. I noticed, that there is inconsistency in returned format, e.g. for given test:
```
module top(input a, output b);
        logic [11:0] first  = 12'h1; //returned format: vpiHexStrVal
        logic [11:0] second  = 12'd1; //returned format: vpiIntVal
        logic [11:0] third  = 12'b1; //returned format: vpiBinStrVal
        logic [11:0] forth  = 12'o1; //returned format: vpiOctStrVal
endmodule
```
only for ``12'b1`` returned format is ``vpiIntVal``, for others ``vpi*StrVal``. I would suspect, that either all of them would return  ``vpiIntVal`` (as a format, not ``vpiConsType``) or ``12'd1`` would also return ``vpiDecStrVal``.

PR #977 (2020-12-19): dec values support
Fixes #967
closed 2020-12-19 by alaindargelas