chipsalliance/Surelog #974: [Windows only] slpp_all/work directories easily exceeds max path length

DamienLeBars, 2020-12-18:
std::filesystem::create_directories throw easily an exception during the slpp_all/work hierarchical directory creation due to windows limit (260). (happends if max length path of input file + output directory path >260)
Maybe, a flat structure with some mapping could reduce this risk.

alaindargelas, 2020-12-19:
Windows has a setting for this problem.
Most commercial EDA tools recommend using the following setting:
https://docs.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation

alaindargelas, 2020-12-19:
The alternative requires coming up with a name mangling mechanism for the cache directory structure. 
I'll happily leave that task to the Windows users (under guidance) ;-)

hs-apotell, 2021-03-08:
Personal opinion - I don't think this should be a windows only feature. In fact, nothing should be platform specific unless otherwise absolutely necessary. Features like these open door for bugs to creep in which happens only on one platform and not on other, thus demanding more dedicated testing of each feature on every supported platform.

On potential solution to the problem is to maintain a mapping from original source path to a UUID (mapped to the file path) and store all files as a single packed file. i.e. index and the original source can be packed into the same file on disk, kind of like a tarball. We can certainly find some open source implementations that can be leveraged.

alaindargelas, 2021-03-08:
Note to potential implementer,Verilog allows for multiple files having the same name as long they reside in different directories.The hashing mechanism needs to account for that and make is easily cross probed when debugging issues (Point back to the original file).

PR #1184 (2021-04-10): Hash preprocessor file paths
Trying to fix #974
closed 2021-04-10 by alaindargelas