chipsalliance/Surelog #982: Static evaluation of constant assignment patterns

kbieganski, 2020-12-22:
Given this Verilog code:
```systemverilog
module ibex_csr #(
    parameter bit [4:0] ResetValue = '0
) (
    input logic i,
    output logic o
);
   assign y = x;
endmodule;

module top();
  logic i;
  logic o;

  typedef struct packed {
    logic mie;
    logic mpie;
    logic mpp;
    logic mprv;
    logic tw;
  } status_t;

  localparam status_t MSTATUS_RST_VAL = '{mie:  1'b1,
                                          mpie: 1'b1,
                                          mpp:  1'b0,
                                          mprv: 1'b0,
                                          tw:   1'b0};
  ibex_csr #(
    .ResetValue ({MSTATUS_RST_VAL})
  ) u_mstatus_csr(.i(i), .o(o));
endmodule
```
Surelog generates the following UHDM (some parts skipped):
```
\_module: work@top (work@top) top.sv:10:
   ...
   \_module: work@ibex_csr (work@top.u_mstatus_csr) top.sv:26: , parent:work@top
     ...
     |vpiParamAssign:
     \_param_assign: , line:2, parent:work@top.u_mstatus_csr
       |vpiRhs:
       \_constant: , line:2
         |vpiConstType:7
         |vpiDecompile:0
         |vpiSize:1
         |SCAL:0
       |vpiLhs:
       \_parameter: (work@ibex_csr.ResetValue), line:2, parent:work@ibex_csr
     |vpiParameter:
     \_parameter: (work@top.u_mstatus_csr.ResetValue), line:2, parent:work@top.u_mstatus_csr
       |vpiName:ResetValue
       |vpiFullName:work@top.u_mstatus_csr.ResetValue
       |SCAL:0
```
The parameter is simply assigned 0 instead of the value from the assignment pattern.

This is because at https://github.com/alainmarcel/Surelog/blob/6a0bca090a30f929f16a9e4999e67ccf70a7b366/src/DesignCompile/DesignElaboration.cpp#L1326 the value for `MSTATUS_RST_VAL` is being set to 0, due to `ExprBuilder::evalExpr` not supporting `slConstant_assignment_pattern_expression`s.

kbieganski, 2020-12-22:
I'm trying to add support for this, but it seems non trivial. So far I have this: https://github.com/antmicro/Surelog/commit/b6fbf54aa9022cc5b2bc99f171bded1d3ebf5b24
which **is not** ready for a PR, due to only working in a specific case (when the values are listed in the correct order – it ignores the pattern keys).

kgugala, 2020-12-22:
@alaindargelas please take a look at this one

kbieganski, 2020-12-22:
Real world example: https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv#L745

PR #986 (2020-12-23): param with complex values
Fixes #982
closed 2020-12-23 by alaindargelas