chipsalliance/Surelog #1001: Complex type parameter nodes are missing values

rkapuscik, 2021-01-07:
A test case to consider for parameter evaluation (adapted from https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv): 
```verilog
   localparam pkg::complex_t init_val = '{a: 1'b0, b: '1};
   dut #(
       .size ($bits(pkg::complex_t)),
       .init ({init_val})
     ) asgn0(
       .inp(i),
       .out(o1)
     );
```
Full test case: https://github.com/antmicro/uhdm-integration/blob/ibex-cs-registers/tests/ComplexParam/top.sv
Both `size` and `init` vpiParameter nodes have no value in UHDM.

alaindargelas, 2021-01-10:
I create a simplified testcase:
https://github.com/alainmarcel/Surelog/tree/master/tests/ParamComplex
Note that it has all the correct values, maybe this got fixed recently.

rkapuscik, 2021-01-11:
Thanks for looking at this, but there are still two problems here:

1. The values are stored in the vpiParamAssign nodes, vpiParameter nodes do not have any values in this test case
2. Surelog strips one layer of indirection in these parameters, but does not provide the final values: notice that it would be possible to skip the localparam and provide the pattern as parameter value, but e.g. Verilator does not allow that (pattern under pin node). Surelog does not allow this workaround.

alaindargelas, 2021-01-17:
This feature: "The values are stored in the vpiParamAssign nodes, vpiParameter nodes do not have any values in this test case" 
is more formalized and is described in #1020

closed 2021-01-17 by alaindargelas