chipsalliance/Surelog #1010: Imported parameter nodes are indistinguishable

rkapuscik, 2021-01-12:
Consider another modified https://github.com/alainmarcel/uhdm-integration/blob/master/tests/MultipleCells/top.sv, where the modules import a package that contains a parameter:
```verilog
package param_pkg;
  parameter logic PKG_PARAM = 0;
  localparam logic PKG_LOCALPARAM = 1;
endpackage

module top (input wire i, output reg o1, output reg o2);
   import param_pkg::*;
   assigner #(.invert(0)) asgn0(.inp(i), .out(o1));
endmodule

module assigner #(parameter invert = 0) (input wire inp, output reg out);
  import param_pkg::*;
  if (!invert)
    assign out = inp;
  if (invert)
    assign out = ~inp;
endmodule  
```
Now in UHDM:

1. there are separate nodes for PKG_PARAM and PKG_LOCALPARAM in both `assigner` and `top` modules in `uhdmTopModules` (this is not a problem in itself, but could be misleading)
2. there is no field/information (and I don't actually see one in the standard) that would differentiate type of the imported `PKG_PARAM` node from the module parameter `invert` - they are both accessed through `vpiParameter` or `vpiParamAssign` calls, but only the `invert` should be passed when the module is instantiated.

This becomes even more confusing when the module has its own parameters that are used with their default values and not set when creating the instance. I think a field to indicate whether a parameter was imported (or where it came from) would be useful, but I see no such distinction in the standard.

alaindargelas, 2021-01-13:
A VpiImported boolean as a property for parameters would do the trick?

rkapuscik, 2021-01-13:
It would. I wonder if this would be necessary if the parameters were stored in a consistent way, as we discussed. But since parameter assignments can get really complicated and as you said in https://github.com/alainmarcel/Surelog/issues/1013#issuecomment-759266631 it would be hard to compute all values for vpiParameter nodes, something like VpiImported would be needed for distinction.

alaindargelas, 2021-01-13:
I'll make is a string type,  imported parameters will carry the name of the package they got imported from.

alaindargelas, 2021-01-17:
Fixed by #1020

closed 2021-01-17 by alaindargelas