chipsalliance/Surelog #1013: Indirection removed from parameters without providing final values

rkapuscik, 2021-01-13:
Split from #1001
 ```verilog
  localparam pkg::complex_t init_val = '{a: 1'b0, b: '1};
   dut #(
       .size ($bits(pkg::complex_t)),
       .init ({init_val})
     ) asgn0(
       .inp(i),
       .out(o1)
     );
```
Full test case: https://github.com/antmicro/uhdm-integration/blob/ibex-cs-registers/tests/ComplexParam/top.sv

The reason for this localparam is that Verilator does not allow a pattern to be assigned to parameter when creating module instance (under pin node). Surelog strips one layer of indirection in these parameters, but does not provide the final values, disabling the workaround.

alaindargelas, 2021-01-13:
I don't understand that statement:
"Surelog strips one layer of indirection in these parameters", can you use Verilog or point to a faulty UHDM model  and what you want to see instead to explain what you mean? But I'm not sure it is important.
I think what you are asking is that Surelog computes: {'{a: 1'b0, b: '1}} -> 0001 for you ?

alaindargelas, 2021-01-13:
If that is the case, I'm afraid Surelog is missing a pass to to that and that it is equally easy/difficult to implement that in Surelog prior to create UHDM or when you retrive UHDM and compute the scalar in you UHDM->Veriator code

rkapuscik, 2021-01-13:
I'll try to explain that in UHDM - for the assignment to .init in dut:
```
dut #(
      .size ($bits(pkg::complex_t)), // ignore this one in this issue
      .init ({init_val})
    ) asgn0(
```
Created UHDM (uhdmtopModules view)
```
    |vpiParamAssign:
    \_param_assign: , line:27, parent:work@top.asgn0
      |vpiRhs:
      \_operation: , line:18
        |vpiOpType:33
        |vpiOperand:
        \_operation: , line:18
          |vpiOpType:75
          |vpiOperand:
          \_tagged_pattern: 
            |vpiTypespec:
            \_string_typespec: (a), line:15
              |vpiName:a
            |vpiPattern:
            \_constant: , line:15
              |vpiConstType:3
              |vpiDecompile:1'b0
              |vpiSize:1
              |BIN:0
          |vpiOperand:
          \_tagged_pattern: 
            |vpiTypespec:
            \_string_typespec: (b), line:15
              |vpiName:b
            |vpiPattern:
            \_constant: , line:15
              |vpiConstType:3
              |vpiDecompile:'b1
              |BIN:1
      |vpiLhs:
      \_parameter: (work@dut.init), line:27, parent:work@dut
        |vpiName:init
        |vpiFullName:work@dut.init
        |vpiTypespec:
        \_bit_typespec: (init), line:27, parent:work@dut.init
          |vpiName:init
          |vpiRange:
          \_range: , line:27
            |vpiLeftRange:
            \_operation: , line:27
              |vpiOpType:11
              |vpiOperand:
              \_ref_obj: (size), line:27
                |vpiName:size
              |vpiOperand:
              \_constant: , line:27
                |vpiConstType:7
                |vpiDecompile:1
                |vpiSize:32
                |INT:1
            |vpiRightRange:
            \_constant: , line:27
              |vpiConstType:7
              |vpiDecompile:0
              |vpiSize:32
              |INT:0
    |vpiParameter:
    \_parameter: (work@top.asgn0.init), line:18, parent:work@top.asgn0
      |vpiName:init
      |vpiFullName:work@top.asgn0.init
```
So Surelog strips the local parameter `init_val` and stores the pattern in parameter assignment. This is not allowed in Verilator, which is why the localparam was used in the first place.
I see two possible solutions:
1. Store the final value of the parameter in the UHDM (complicated and not really necessary here)
2. Keep the localparam - in this case assign the pattern to init_val (this is done already), when creating the instance keep the reference to init_val.
3. (Fix the reason of the error in Verilator, but that's beyond the scope for now since Ibex uses this workaround)

For reference, Verilator frontend creates following structure (similar to 2. above):
```
    1:2: VAR 0x55bd703be580 <e126#> {c15be} @dt=0@  init_val LPARAM
    1:2:1: REFDTYPE 0x55bd70418300 <e80#> {c15au} @dt=0@  complex_t -> UNLINKED
    1:2:1:3: CLASSORPACKAGEREF 0x55bd703e08c0 <e59#> {c15ap}  pkg cpkg=0x55bd70436000 -> PACKAGE 0x55bd70436000 <e206#> {c1aj}  pkg  L3 [LIB] [1ps]
    1:2:3: PATTERN 0x55bd70442300 <e81#> {c15bp} @dt=0@
    1:2:3:2: PATMEMBER 0x55bd70442180 <e66#> {c15br} @dt=0@
    1:2:3:2:1: CONST 0x55bd7043a7e0 <e65#> {c15bu} @dt=0x55bd7043e8f0@(G/w1)  1'h0
    1:2:3:2:2: TEXT 0x55bd703e0700 <e67#> {c15br} "a"
    1:2:3:2: PATMEMBER 0x55bd70442240 <e77#> {c15ca} @dt=0@
    1:2:3:2:1: CONST 0x55bd7043a900 <e74#> {c15cd} @dt=0x55bd7043e8f0@(G/w1)  '1
    1:2:3:2:2: TEXT 0x55bd703e07e0 <e76#> {c15ca} "b"
    1:2: CELL 0x55bd70436280 <e125#> {c19ai}  asgn0 -> MODULE 0x55bd704363c0 <e207#> {c25ai}  dut  L3 [1ps]
    (... other ports and parameters)
    1:2:2: PIN 0x55bd704465a0 <e104#> {c18aj}  init ->UNLINKED
    1:2:2:1: REPLICATE 0x55bd70442600 <e103#> {c18ap} @dt=0x55bd7043eb60@(G/w0)
    1:2:2:1:1: PARSEREF 0x55bd703e0e00 <e96#> {c18aq}  init_val [TEXT]
    1:2:2:1:2: CONST 0x55bd7043ab40 <e97#> {c18ap} @dt=0x55bd7043ea90@(G/w32)  32'h1

```

alaindargelas, 2021-01-13:
I get it now when you said: "...strips the local parameter" you meant "...**substitute** the local parameter identifier by its value".
Also, did you miss these lines:
\_operation: , line:18
        |vpiOpType:33
This is a concatenation operator. So you could perform the concatenation operation and reduce the subtree to a scalar.

PR #1021 (2021-01-17): Fix for Paramater Verilator limitation
Fixes #1013

@rkapuscik, please add -verilator to the Surelog command line or use the SURELOG::CommandLineParser API to set the VerilatorMode on.

This is a limitation for Verilator that I don't want to make default.
closed 2021-01-17 by alaindargelas