chipsalliance/Surelog #1019: antlr fork

hs-apotell, 2021-01-16:
Surelog includes a copy of the antlr repo under its third party folder. It is our understanding that this is a modified version of the original to introduce some sort of optimizations. Are you planning on pushing these changes back to the original repo? If not, can this source be an actual fork and be included as a submodule (like UHDM and few others) in Surelog.

Reasoning - We are using antlr4 to build other projects and having two different versions of the library creates conflict. The other project isn't dependent on Surelog and introducing this dependency is not ideal and in some cases cause circular dependencies. Pulling it out as a separate fork would allow us to use the same version across multiple projects.

alaindargelas, 2021-01-16:
Sure, can you take that on? I am not going to push these changes on the original repo.
So making it a fork and including it sounds ok.

alaindargelas, 2021-01-16:
Make sure the default version pulled by Surelog is the forked version which is almost 2x faster.

hs-apotell, 2021-01-16:
> can you take that on?

Sure. I will fork the newest version of antlr4 repo and apply the changes in Surelog on top of it. Once ready I can pass on the ownership of the fork on to you.

mithro, 2021-01-16:
@hs-apotell - I believe @hzeller was interested in looking at getting @alaindargelas's antlr changes upstream and also had some other ideas on how to significantly improve performance. Sadly he hasn't had the time to actually do this work at the moment and I'm unsure if he will have time soon. There being a simple and clear way to understand the differences in the antlr used by surelog and upstream antlr is a great goal!

hzeller, 2021-01-19:
I think we should push the changes upstream that from what I remember reducing some dynamic casting situations. It is tentatively on my TODO list, but, well that list is long.
So if you could work on upstreaming the Antlr changes (after confirming that they are safe in all cicumstances) that would actually be great so that everyone can benefit from it and you require less effort on maintaining the branch.

As a first step, doing a fork and applying the patches there to be then used as a submodule sounds like a good start.

alaindargelas, 2021-01-27:
Fixed by #1040, thanks @hs-apotell !

closed 2021-01-27 by alaindargelas