chipsalliance/Surelog #1035: Wrong real parameter value in uhdmtopModules

kamilrakoczy, 2021-01-22:
TEST:
```
module top(input clk_i, output b);

  dut #(.CLKFBOUT_PHASE(12.50)) d (
  .a(clk_i),
  .b(b));
endmodule

module dut #(
  parameter real CLKFBOUT_PHASE = 1142.500
) (input a, output b);
  assign a = b;
endmodule
```

UHDM:
```
|uhdmallModules:                                                                                                                                                                                       
 \_module: work@dut (work@dut) /home/grey/uhdm/uhdm-integration/tests/t/top.sv:8: , parent:work@top                                                                                                                
   |vpiDefName:work@du
   ...
   |vpiParamAssign:                                                                                                                                                                                                
   \_param_assign: , line:9, parent:work@dut                                                                                                                                                                       
     |vpiRhs:                                                                                                                                                                                                      
     \_constant: , line:9                                                                                                                                                                                          
       |vpiConstType:2                                                                                                                                                                                             
       |vpiDecompile:1142.500                                                                                                                                                                                      
       |REAL:1142.500000                                                                                                                                                                                           
     |vpiLhs:
     \_parameter: (work@dut.CLKFBOUT_PHASE), line:9, parent:work@dut
       |vpiName:CLKFBOUT_PHASE
       |vpiFullName:work@dut.CLKFBOUT_PHASE
       |vpiTypespec:
       \_real_typespec: (CLKFBOUT_PHASE), line:9, parent:work@dut.CLKFBOUT_PHASE
         |vpiName:CLKFBOUT_PHASE
...
 |uhdmtopModules:                                                                                                                                                                                                  
 \_module: work@top (work@top) /home/grey/uhdm/uhdm-integration/tests/t/top.sv:1:                                                                                                                                  
   |vpiDefName:work@top 
...
     |vpiParamAssign:                                                                                                                                                                                              
     \_param_assign: , line:9, parent:work@top.d                                                                                                                                                                   
       |vpiRhs:                                     
       \_constant: , line:9                                                                              
         |vpiConstType:7                            
         |vpiDecompile:12                                                                                
         |vpiSize:64             
         |INT:12                                                                                         
       |vpiLhs:                                                                                          
       \_parameter: (work@top.d.CLKFBOUT_PHASE), line:9, parent:work@top.d                               
         |vpiName:CLKFBOUT_PHASE                                                                         
         |vpiFullName:work@top.d.CLKFBOUT_PHASE                                                                                                                                                                    
         |INT:12                                                                                                                                                                                                   
         |vpiTypespec:                                                                                   
         \_real_typespec: (CLKFBOUT_PHASE), line:9, parent:work@top.d.CLKFBOUT_PHASE                     
           |vpiName:CLKFBOUT_PHASE
```
Real parameters value reported in ``uhdmallModules`` are as expected, but one in ``uhdmtopModules`` is reported as a ``INT`` (even when we are not overwriting it in top module).

alaindargelas, 2021-01-23:
Reals are not guaranteed/tested in the elaboration process at this point

alaindargelas, 2021-03-11:
Fixed by #1110

closed 2021-03-11 by alaindargelas