chipsalliance/Surelog #1036: Missing way of detecting if module is a library

kamilrakoczy, 2021-01-22:
Both ``verilator`` and ``yosys`` have special way of treating with libraries.
Surelog can read library file using:
```
-v <file>             Library file
-y <path>             Library directory
```
but there is no way of detecting (in UHDM) if module is a library, or not. In both cases (when file is read with ``-v/-y`` flag or as a regular system verilog file)  surelog produces exactly the same UHDM for library module.

Example test:
```
module PLLE2_ADV ();
    parameter BANDWIDTH = "OPTIMIZED";
    parameter COMPENSATION = "ZHOLD";
    parameter STARTUP_WAIT = "FALSE";
    output CLKFBOUT;
    output CLKOUT0;
    output CLKOUT1;
    output CLKOUT2;
    output CLKOUT3;
    output CLKOUT4;
    output CLKOUT5;
    output LOCKED;
    input CLKFBIN;
    input CLKIN1;
    input PWRDWN;
    input RST;
endmodule
```
Library cmd: ``surelog -parse -sverilog -d uhdm -v tests/t/lib.sv``
Regular sv cmd: ``surelog -parse -sverilog -d uhdm tests/t/lib.sv``

alaindargelas, 2021-01-23:
That is because the standard VPI does not offer such distinction either.
Is this important for some specific reason?

kamilrakoczy, 2021-01-26:
``yosys`` expects that library modules will have ``lib_blackblox``/``lib_whitebox`` attribute. Now, I was checking if module is a blacbox (have only wires that are input/output, parameters or cells) and then marking it as a library.
But it is not always the case, e.g.:
```
module BUFG(
    (* clkbuf_driver *)
    output O,
    input I);
  assign O = I;
  specify
    // https://github.com/SymbiFlow/prjxray-db/blob/4bc6385ab300b1819848371f508185f57b649a0e/artix7/timings/CLK_BUFG_TOP_R.sdf#L11
    (I => O) = 96;
  endspecify
endmodule
```
Without this attribute, I'm removing this module (and corresponding cell) from the design, as I do not detect it is a library module.
Bitstream generated by ``vivado`` is working without this cell, but it is not the case for ``symbiflow-toolchain``

alaindargelas, 2021-01-27:
Correction, the standard provides for this: vpiCellInstance

PR #1042 (2021-01-27): vpiCellInstance
Fixes #1036
closed 2021-01-27 by alaindargelas