chipsalliance/Surelog #1084: Missing vpiNet for custom type variables in vpiGenScope

kamilrakoczy, 2021-02-25:
Test:
```
package ibex_pkg;
typedef enum logic[1:0] {
  WB_INSTR_LOAD,
  WB_INSTR_STORE,
  WB_INSTR_OTHER
} wb_instr_type_e;
endpackage

module dut();
import ibex_pkg::*;
parameter bit TEST = 1'b1;
logic out;
wb_instr_type_e first_var;
if(TEST) begin
  wb_instr_type_e second_var;
  logic           third_var;
  assign first_var = WB_INSTR_LOAD;
  assign second_var = WB_INSTR_STORE;
  assign third_var = 1'b1;
  assign out = 1'b1;
end else begin
  assign out = 1'b0;
end

endmodule
```
UHDM:
```
\_gen_scope: (work@dut.genblk1), parent:work@dut.genblk1                                                                                                                                                      
       |vpiFullName:work@dut.genblk1                                                                                                                                                                               
       |vpiNet:                                                                                                                                                                                                    
       \_logic_net: (work@dut.genblk1.third_var), line:16, parent:work@dut.genblk1                                                                                                                                 
         |vpiName:third_var                                                                                                                                                                                        
         |vpiFullName:work@dut.genblk1.third_var                                                                                                                                                                   
         |vpiNetType:36  
```
UHDM contains information only about non-custom type variables declared inside generate scope. It is missing vpiNet for ``second_var`` that is custom type variable declared inside generate scope.

alaindargelas, 2021-03-03:
It's in the variables section:

```
    |vpiVariables:
      \_enum_var: (work@dut.genblk1.second_var), id:35, line:13, parent:work@dut.genblk1, parID:45
        |vpiName:second_var
        |vpiFullName:work@dut.genblk1.second_var
        |vpiAutomatic:1
        |vpiVisibility:1
        |vpiTypespec:
        \_enum_typespec: (wb_instr_type_e), id:6, line:4
          |vpiName:wb_instr_type_e
          |vpiBaseTypespec:
          \_logic_typespec: , id:5, line:2
            |vpiRange:
            \_range: , id:2, line:2
              |vpiLeftRange:
              \_constant: , id:3, line:2, parID:2
                |vpiConstType:7
                |vpiDecompile:1
                |vpiSize:64
```

alaindargelas, 2021-03-03:
Please close the issue if that is acceptable

kamilrakoczy, 2021-03-03:
Yes, this is acceptable. Closing issue.

closed 2021-03-03 by kamilrakoczy