chipsalliance/Surelog #1104: Rerouting logs

hs-apotell, 2021-03-08:
Current implementation writes all log statements to dedicated log files. We would like to change this behavior and like the logs be redirected to some sort of a listener. We have a dedicated log window for our tool where we would like the warnings/errors to be accessible. The current behavior will be the default and will be maintained (except in the case where the behavior gets overwritten by setting a different listener).

I am in the process of implementing this enhancement. If you any guidance on how you would like this to work, please share.

alaindargelas, 2021-03-10:
All messages already go through a logging mechanism. They can be muted from stdout, and the log file can be redirected to any location (See command line options). Adding a 3rd stream (ostream) registered with a callback should be trivial in the existing logging mechanism.

hs-apotell, 2021-03-10:
I don't think `ostream` based redirection would require polling, which isn't ideal. Plus, it introduces unnecessary synchronization issues if the parser was run on background thread.

I was intending for it to be a callback. A simple interface that can be implemented outside the core Surelog implementation and be registered. The existing `ostream` based mechanism can be wrapped inside of a default implementation of the interface i.e. within Surelog.

closed 2021-03-13 by hs-apotell