chipsalliance/Surelog #1109: Missing information about imported from package functions

kamilrakoczy, 2021-03-11:
Test:
```
package pkg;
function automatic logic [31:0] package_func(logic [31:0] in);
  logic [31:0] out;
  assign out = in;
  return out;
endfunction

typedef enum logic [6:0] {
  TEST_ENUM_VAL     = 7'h03
} package_enum;
endpackage

module dut();
import pkg::*;

logic      [31:0] in, out;

assign in = 32'h12345678;

assign out = package_func(in);
endmodule
```
UHDM:
```                                                                                                                                                                                                                                                                                                
   |vpiDefName:work@dut                                                                                                                                                                                            
   |vpiName:work@dut                                                                                                                                                                                               
   |vpiNet:                                                                                                                                                                                                        
   \_logic_net: (work@dut.in), line:17, parent:work@dut                                                                                                                                                            
   |vpiNet:                                                                                                                                                                                                        
   \_logic_net: (work@dut.out), line:17, parent:work@dut                                                                                                                                                           
     |vpiName:out                                                                                                                                                                                                  
     |vpiFullName:work@dut.out                                                                                                                                                                                     
     |vpiNetType:36                                                                                                                                                                                                
     |vpiRange:                                                                                                                                                                                                    
     \_range: , line:17                                                                                                                                                                                            
       |vpiLeftRange:                                                                                                                                                                                              
       \_constant: , line:17                                                                                                                                                                                       
         |vpiConstType:7                                                                                                                                                                                           
         |vpiDecompile:31                                                                                                                                                                                          
         |vpiSize:64                                                                                                                                                                                               
         |INT:31                                                                                                                                                                                                   
       |vpiRightRange:                                                                                                                                                                                             
       \_constant: , line:17                                                                                                                                                                                       
         |vpiConstType:7                                                                                                                                                                                           
         |vpiDecompile:0                                                                                                                                                                                           
         |vpiSize:64                                                                                                                                                                                               
         |INT:0                                                                                                                                                                                                    
   |vpiTypedef:                                                                                                                                                                                                    
   \_enum_typespec: (package_enum), line:10                                                                                                                                                                        
   |vpiTypedef:                                                                                                                                                                                                    
   \_import: (pkg)
```

In case of imported items, UHDM is containing information only about typedefs, but it is missing information about e.g. functions.

UHDM is containing information about import of ``pkg`` where used function is defined, but I can't import this package in yosys, as some information from this package (in this case enum typedef) is also present in the module definition.

There 2 possible solutions to this problem:
1. add information about used imported functions to the module
2. remove information about imported typedefs and left importing of package to yosys.

alaindargelas, 2021-03-11:
The only viable choice is:
1. add information about used imported functions to the module

alaindargelas, 2021-03-11:
Actually, better there is a 3rd choice:

When you detect an imported typedef:

```
|vpiTypedef:
        \_enum_typespec: (package_enum), id:27, line:12
          |vpiName:package_enum
          |vpiBaseTypespec:
          \_logic_typespec: , id:26, line:10
            |vpiRange:
            \_range: , id:23, line:10
              |vpiLeftRange:
              \_constant: , id:24, line:10, parID:23
                |vpiConstType:9
                |vpiDecompile:6
                |vpiSize:64
                |UINT:6
              |vpiRightRange:
              \_constant: , id:25, line:10, parID:23
                |vpiConstType:9
                |vpiDecompile:0
                |vpiSize:64
                |UINT:0
            |vpiInstance:
            \_package: pkg (pkg), id:1 tests/ScratchPad.sv:3: , parent:work@dut, parID:46
```
using the vpiInstance field of the typedef: 

```
 |vpiInstance:
            \_package: pkg (pkg), id:1 tests/ScratchPad.sv:3: , parent:work@dut, parID:46
```

then you can decide to not import it in Yosys, does that work?

kamilrakoczy, 2021-03-11:
I think, I could skip creating typedef node, if I would found ``vpiInstance`` field. Then I would need to import whole package in the yosys. 

This looks very similar to 2nd option (as we skipping typedefs in module, that are imported), but implementation is moved to the yosys part. I think very similar thing will be needed for the verilator part as well, so I'm not sure it is best way.

alaindargelas, 2021-03-11:
I can't remove the information from imported typedefs, I need it in UHDM itself to produce the complete UHDM.

alaindargelas, 2021-03-11:
There are several things that are imported (inline) from packages:
- variables
- functions
- typedefs
- constants created by enum

All could be inline in the importing module, right now only variables, typedefs and constants are.

Would you prefer that all are inline and have the vpiInstance field set?

kamilrakoczy, 2021-03-11:
Yes, inlining all of them should be enough for yosys and verilator and should solve this issue for both tools.

alaindargelas, 2021-03-12:
Fixed by #1114

closed 2021-03-12 by alaindargelas