chipsalliance/Surelog #1112: Regression after

kamilrakoczy, 2021-03-11:


alaindargelas, 2021-03-12:
The declaration type of the variable is of no matter here.
What matters is the actual value. The value 10 is a positive integer, and 0 can be encoded as an unsigned integer too, hence these values are encoded as unsigned. This is to fix an issue with NAN or -NAN.
If you need to satisfy Yosys by fudging with the data type, to make it happy, please do so, it is a small price to pay to be able to encode the values correctly.

alaindargelas, 2021-03-12:
Moreover you have the type of the declaration to be int, if you see the cast is safe, you can cast the rhs to the lhs type.

kamilrakoczy, 2021-03-12:
Another test:
```
module dut();
integer s;

assign s = -1;

endmodule
```
UHDM:
```
|vpiContAssign:                                                                                                                                                                                                 
   \_cont_assign: , line:73, parent:work@dut                                                                                                                                                                       
     |vpiRhs:                                                                                                                                                                                                      
     \_operation: , line:73                                                                                                                                                                                        
       |vpiOpType:1                                                                                                                                                                                                
       |vpiOperand:                                                                                                                                                                                                
       \_constant: , line:73                                                                                                                                                                                       
         |vpiConstType:9                                                                                                                                                                                           
         |vpiDecompile:1                                                                                                                                                                                           
         |vpiSize:64                                                                                                                                                                                               
     |vpiLhs:                                                                                            
     \_ref_obj: (work@dut.s), line:73                                                                    
       |vpiName:s                                                                                                                                                                                                  
       |vpiFullName:work@dut.s       
 ```
 Here UHDM also reports ``-1`` as ``|vpiConstType:9`` and even changes its value to ``1`` from ``-1``, so the actual value is wrong here.
 
 I can't easily cast rhs signed/unsigned type to lhs type, as I do not have connection between ``wire`` (where signed type is saved) and ``identifier`` (``ref_obj``) until AST representation (so I would need to change internals of yosys, I can't do that just for uhdm frontend).

alaindargelas, 2021-03-12:
Is this real code from a real test case?Surelog preserve negative numbers on parameters only.

Reopened #1112.

alaindargelas, 2021-03-12:
As a matter of fact, this is not a regression, this was never working, the lexer/parser drops the negative sign in the first place.
UHDM never sees that negative value.

closed 2021-03-12 by alaindargelas