chipsalliance/Surelog #1120: Missing column information in parsed data

hs-apotell, 2021-03-14:
UHDM::BaseClass provides access to File name and Line No. but the column number is not getting propagated to it from the parser.

Food for thought - Would it be worth (or at least an option to) carrying forward the actual token data i.e. the ANTLR primitive forward to allow access to other critical information for the purposes of linting. Being able to tie the model back to actual token can be useful.

alaindargelas, 2021-04-01:
#1165 provides persisted column info. It is not the solutions discussed above, but should permit many applications.

alaindargelas, 2021-04-05:
#1170 introduces all the necessary infra to save that data in UHDM.
The changelist only saves the module information. All other objects of interest have to have their endline and endcolumn added to UHDM from the internal AST.

alaindargelas, 2021-04-11:
Fixed by #1197

closed 2021-04-11 by alaindargelas

hs-apotell, 2021-04-25:
> The changelist only saves the module information. All other objects of interest have to have their endline and endcolumn added to UHDM from the internal AST.

@alaindargelas Quick follow up. I just pulled down the latest and looks like you already completed the pending work. Is there anything left on this task to do?

alaindargelas, 2021-04-25:
Nope, it's all done. The issue is closed.