chipsalliance/Surelog #1129: Missing vpiInstance in cells created in generate scope

kamilrakoczy, 2021-03-17:
Test:
```
 module test_module();
 endmodule
 
 module dut();
   for (genvar i = 0; i < 4; i++) begin : gen_mix_column
     test_module test_module();
   end
 endmodule
```
UHDM:
```
        |vpiModule:
        \_module: work@test_module (work@dut.gen_mix_column[0].test_module) /home/grey/uhdm/yosys-tests/1/yosys/uhdm-integration/tests/t/dut.sv:6: , parent:work@dut.gen_mix_column[0]
          |vpiDefName:work@test_module
          |vpiName:test_module
          |vpiFullName:work@dut.gen_mix_column[0].test_module
```
There missing ``vpiInstance``. This information is useful to distinguish between module definition and cell that has the same name as the module. This information is present, when cell is not generated in generate scope.:
Test:
```
module test_module();
endmodule
   
module dut();
    test_module test_module();
endmodule
```
```
    |vpiModule:
    \_module: work@test_module (work@dut.test_module) /home/grey/uhdm/yosys-tests/1/yosys/uhdm-integration/tests/t/dut.sv:5: , parent:work@dut
      |vpiDefName:work@test_module
      |vpiName:test_module
      |vpiFullName:work@dut.test_module
      |vpiInstance:
      \_module: work@dut (work@dut) /home/grey/uhdm/yosys-tests/1/yosys/uhdm-integration/tests/t/dut.sv:4:
```

alaindargelas, 2021-03-18:
This relation does not exist in the VPI Standard Object Model, we would have to add it which is possible, 
but first, why can't you use the vpiParent relation and the vpiDefName and vpiName to achieve that distinction?

What you refer as a "cell" is an instance in Verilog lingo, a "cell" is actually typically referring to a Gate level cell primitive.

Moreover:
- the module definition name (vpiDefName) always contains the name of the library (Here: work@...) where 
- the module instance name (vpiName) contains the instance name as in the RTL text (not prefixed with the lib name).

kamilrakoczy, 2021-03-18:
Looks like this comparison is enough, so there is no need to add this relation if it is not in VPI Standard Object Model. Closing issue.

closed 2021-03-18 by kamilrakoczy