chipsalliance/Surelog #1147: Bit values aren't parsed properly when passed by -P flag

RRozak, 2021-03-26:
Hi, 
I want to overwrite the default value of the bit parameter. I noticed that when bit value is passed it is taken as int.
Here is a simple test:
```systemverilog
module top();
    parameter p = 1'b0;
endmodule
```
Here is the uhdm tree:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) bit_parameter.sv:6: , parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiParamAssign:
  \_param_assign: , line:7, parent:work@top
    |vpiRhs:
    \_constant: , line:7
      |vpiConstType:3
      |vpiDecompile:1'b0
      |vpiSize:1
      |BIN:0
    |vpiLhs:
    \_parameter: (work@top.p), line:7, parent:work@top
      |vpiName:p
      |vpiFullName:work@top.p
  |vpiParameter:
  \_parameter: (work@top.p), line:7, parent:work@top
|uhdmtopModules:
\_module: work@top (work@top) bit_parameter.sv:6: 
  |vpiDefName:work@top
  |vpiName:work@top
  |vpiParamAssign:
  \_param_assign: , line:7, parent:work@top
    |vpiRhs:
    \_constant: , line:7
      |vpiConstType:9
      |vpiDecompile:1
      |vpiSize:64
      |UINT:1
    |vpiLhs:
    \_parameter: (work@top.p), line:7, parent:work@top
      |vpiName:p
      |vpiFullName:work@top.p
      |UINT:1
  |vpiParameter:
  \_parameter: (work@top.p), line:7, parent:work@top
```
I use the following command to run Surelog:
`surelog -parse -nobuiltin -Pp=1\'b1 bit_parameter.sv`

When I changed that value to `2\b'1` it was parsed as `2`.

alaindargelas, 2021-03-26:
Correct command line value parsing is currently very primitive, will add full support.

closed 2021-03-27 by alaindargelas

RRozak, 2021-03-29:
Now it works. Thank you.

PR #1151 (2021-03-27): param setting from cmd line
Fixes #1147