chipsalliance/Surelog #1164: In VpiListener, enterCont_assign called twice.

bsp13, 2021-04-01:
This function called twice...
```C++
virtual void enterCont_assign(
      const UHDM::cont_assign* object, const UHDM::BaseClass* parent,
      vpiHandle handle, vpiHandle parentHandle) 
```

```
module inverter(
	input wire a,
	output wire b
	);
assign b=~a;   ---> called twice
endmodule
```

alaindargelas, 2021-04-03:
That is correct, the module is visited once part of the flat list of modules, then another time part of the hierarchy traversal:
vpi_listener.cpp
```
itr = vpi_iterate(uhdmallModules,object);
    if (itr) {
      while (vpiHandle obj = vpi_scan(itr) ) {
        listen_module (obj, listener);
        vpi_free_object(obj);
      }
      vpi_free_object(itr);
    }
    itr = vpi_iterate(uhdmtopModules,object);
    if (itr) {
      while (vpiHandle obj = vpi_scan(itr) ) {
        listen_module (obj, listener);
        vpi_free_object(obj);
      }
      vpi_free_object(itr);
    }
```

You can distinguish the 2 passes by listening to the module:

```
 void enterModule(const module* object, const BaseClass* parent,
                   vpiHandle handle, vpiHandle parentHandle) 
```

then doing the following (like in ElaboratorListener.h):

```
 module* inst = (module*) object;
    bool topLevelModule         = object->VpiTopModule();
    const std::string& instName = object->VpiName(); 
    const std::string& defName  = object->VpiDefName(); 
    bool flatModule             = (instName == "") && ((object->VpiParent() == 0) ||
                                                       ((object->VpiParent() != 0) && (object->VpiParent()->VpiType() != vpiModule)));
                                  // false when it is a module in a hierachy tree
    if (debug_)
      std::cout << "Module: " << defName << " (" << instName << ") Flat:"  << flatModule << ", Top:" << topLevelModule << std::endl;

    if (flatModule) {
      // Flat list of module (unelaborated)
      flatComponentMap_.insert(std::make_pair(object->VpiDefName(), object));
    } else {
      // Hierachical module list (elaborated)
      inHierarchy_ = true;

```

If you are only interested in the elaborated pass, key on the inHierarchy_ flag that you keep in your Listener.

closed 2021-04-03 by alaindargelas

bsp13, 2021-04-14:
I just wrote log in enterCont_assign function and I got output like this. It is called twice when topmodules are listening.  

```C++
  virtual void enterCont_assign(
      const UHDM::cont_assign* object, const UHDM::BaseClass* parent,
      vpiHandle handle, vpiHandle parentHandle) 
  {
      std::cout << "enter cont assign" << std::endl;
  }
```
OutPut:: 
```
enter cont assign
Entering Module Instance: work@inverter
enter cont assign
enter cont assign
Leaving Module Instance: work@inverter
```

alaindargelas, 2021-04-15:
As you can see from:
https://github.com/alainmarcel/UHDM/blob/master/tests/listener_elab.cpp

Execute:
UHDM$ ./build/bin/listener_elab

......
[1] assign o1 = i1
Module: M1 (M1) Flat:0, Top:1
Module: M1 () Flat:1, Top:0
Module: M2 (inst1) Flat:0, Top:0
[2] assign M1.inst.o1 = M1.inst.i1
Module: M2 () Flat:1, Top:0
[3] assign M1.inst.o1 = M1.inst.i1

Each specific assign debug message is only printed once.

I think you are not filtering the flat vs hierarchical pass correctly in your code.
Please analyze  
https://github.com/alainmarcel/UHDM/blob/master/tests/listener_elab.cpp
vs your implementation.

If there was a bug in the listener, we would see duplicate output in that test.

bsp13, 2021-04-18:
Here is my code  to test enterCont_assign function but did not work proper.

```C++
class DesignListener final : public UHDM::VpiListener
{
    virtual void enterModule(
        const UHDM::module* const object, const UHDM::any* const parent,
        vpiHandle handle, vpiHandle parentHandle) override
    {        
        std::cout << "Entering Module Instance: " << intptr_t(object) << " " << object->UhdmId() << std::endl;
    }

    virtual void enterCont_assign(
        const UHDM::cont_assign* object, const UHDM::BaseClass* parent,
        vpiHandle handle, vpiHandle parentHandle)
    {
        std::cout << "enterCont_assign " << intptr_t(object) << " " << object->UhdmId() << std::endl;
    }
};

bool Build(vpiHandle design_handle)
{
  DesignListener listener;
  UHDM::listen_designs({ design_handle }, &listener);

  return true;
}

int main(int argc, char *argv[])
{
  if (argc < 2)
    return 0;

  // Read command line, compile a design, use -parse argument
  int code = 0;
  SURELOG::SymbolTable *const symbolTable = new SURELOG::SymbolTable();
  
  #if defined(UI_ENABLED)
    apt::davenche::SurelogLogListener *const logListener = new apt::davenche::SurelogLogListener;
    SURELOG::ErrorContainer *const errors = new SURELOG::ErrorContainer(symbolTable, logListener);
  #else
    SURELOG::ErrorContainer *const errors = new SURELOG::ErrorContainer(symbolTable);
  #endif
  SURELOG::CommandLineParser *const clp = new SURELOG::CommandLineParser(errors, symbolTable, false, false);
  clp->noPython();
  clp->setCacheAllowed(false);
  clp->setParse(true);
  clp->setwritePpOutput(true);
  clp->setCompile(true);
  clp->setElaborate(true);
  #if defined(UI_ENABLED)
    clp->setMuteStdout();
  #endif

  bool success = clp->parseCommandLine(argc, const_cast<const char **>(argv));
  errors->printMessages(clp->muteStdout());

  vpiHandle the_design = nullptr;
  if (success && (!clp->help()))
  {
    SURELOG::scompiler *compiler = SURELOG::start_compiler(clp);
    the_design = SURELOG::get_uhdm_design(compiler);
    SURELOG::shutdown_compiler(compiler);
    auto stats = errors->getErrorStats();
    code = (!success) | stats.nbFatal | stats.nbSyntax | stats.nbError;
  }
  delete clp;
  delete symbolTable;
  delete errors;

  if (the_design == nullptr)
    return code;

  UHDM::Serializer serializer;
  UHDM::ElaboratorListener listener(&serializer);
  UHDM::listen_designs({ the_design }, &listener);
  
  if (!Build(the_design))
    return -1;
}
```

OutPut::

```
Entering Module Instance: 2739147934080 11
enterCont_assign 2739160302576 3
Entering Module Instance: 2739147934672 18
enterCont_assign 2739160282336 0
enterCont_assign 2739160277760 4
```

alaindargelas, 2021-04-18:
I added a new target to demonstrate that the behavior is correct:
make hellodesign
Will make and execute the new test. The test will shows a regression compare to the default surelog run (That's fine, its a different executable: hellodesign).Now open the new output file: Surelog/build/tests/Inverter/Inverter.log
You'll see the following content:....[  FATAL] : 0[ SYNTAX] : 0[  ERROR] : 0[WARNING] : 2[   NOTE] : 5Entering Module Definition: ***@***.*** 94196010692032 27  Let's skip the cont assigns in the flat traversal and only navigate the ones in the hierarchical tree!Entering Module Definition: ***@***.*** 94196011232864 34Entering Module Instance: ***@***.*** 94196005687168 41Entering Module Instance: ***@***.*** 94196005792624 42  enterCont_assign 94196010673664 44                                       <<<<<<<<<<< CorrectEntering Module Instance: ***@***.*** 94196005791856 43  enterCont_assign 94196010277456 48                                       <<<<<<<<<<<  CorrectEnd design traversal

As you can see, the enterCont_assign is only traversed 2 times in the elaborated tree (I make the design tests/Inverter/dut.sv more complex to illustrate the concepts of hierarchy tree versus flat tree).

If you notice a difference in Windows, then we have a Linux vs Windows issue.

bsp13, 2021-04-20:
This problem is generated by this argument "-elabuhdm".

can you tell me what is difference between this two condition in CommandLineParser.cpp ?

```C++
  void setElaborate(bool val) { m_elaborate = val; }
```

```C++
  } else if (all_arguments[i] == "-elabuhdm") {
      m_elaborate = true;
      m_elabUhdm = true;
    }
```

Function set only variable "m_elaborate"  but argument set two variables like "m_elaborate" and "m_elabUhdm".

alaindargelas, 2021-04-20:
If you look at the hellodesign.cpp file, You'll see that I programmatically set both variables:
|   |

clp->setElaborate(true);     << Surelog creates its internal hierarchy tree, and create UHDM hierarchy treeclp->setElabUhdm(true);    << This flag performs the UHDM Elaboration (Uniquify all nets, variables, functions, assign statements).
The UHDM Elaboration can be done either before saving UHDM to disk or can be done after reloading UHDM from disk.
If you separate your executables in 2 steps, you save disk space (And separate the concerns): - Surelog compiles, then writes UHDM to disk, - Your client application reloads the UHDM db, then elaborates it on the fly using the Elaboration Listener .
If you link surelog.a and uhdm.a in your executable, then you don't need to save UHDM db and perform both Surelog elab and UHDM elab in the same adress space.

 
 

  void setElaborate(bool val) { m_elaborate = val; }  } else if (all_arguments[i] == "-elabuhdm") {

hs-apotell, 2021-04-20:
Does calling `CommandLineParser::setElaborate(true)` eliminate the need for client side explicit elaboration? If not, what's the purpose of the call? 
```
UHDM::Serializer serializer;
UHDM::ElaboratorListener listener(&serializer);
UHDM::listen_designs({ the_design }, &listener);
```

Is there anything in Surelog's design to track whether a design has been elaborated or not? If not, can we add it and make it accessible to the client?

Both `m_elaborate` and `m_elabUhdm` are boolean flags, and set either programmatically or thru' command-line. Both of these variables are being setup before processing any inputs. Why would we end up with duplicates? Are we elaborating twice - once Surelog doing it internally because of the call to `CommandLineParser::setElaborate(true)` and other because we explicitly, on client side, have the above snippet?

alaindargelas, 2021-04-20:
Correct! You are elaborating UHDM twice, got it now.
You should follow the recipe in hellodesign.cpp, It sets the Surelog and UHDM elaborate flags, everything is done in the Surelog code.You don't need to call the code snippet below.
I'll add the flags you mention below so you can check the status of elaboration.I'll also prevent double elaboration (Just in case).

alaindargelas, 2021-04-21:
You can now replace your code by this code below to test if the design was already elaborated:
```
 if (the_design && (!vpi_get(vpiElaborated, the_design))) {
    UHDM::Serializer serializer;
    UHDM::ElaboratorListener* listener = new UHDM::ElaboratorListener(&serializer, true);
    listen_designs({the_design},listener);
  }
```

Bit if you are in the same process as Surelog.a, then you don't need this, simply use:
```
clp->setElaborate(true);     // Surelog creates its internal hierarchy tree, and create UHDM hierarchy tree
clp->setElabUhdm(true);    // This flag performs the UHDM Elaboration (Uniquify all nets, variables, functions, assign statements).
```