chipsalliance/Surelog #1167: Incomplete vpiDecompile for substituted enums

rkapuscik, 2021-04-01:
For the following assignment: https://github.com/antmicro/uhdm-integration/blob/rk/update-ram2p/tests/EnumConcat/dut.v#L16
UHDM:
```
            |vpiStmt:
            \_assignment: , line:16, parent:work@dut.genblk1[0]
              |vpiOpType:82
              |vpiBlocking:1
              |vpiLhs:
              \_bit_select: (work@dut.genblk1[0].c), line:16
                |vpiName:c
                |vpiFullName:work@dut.genblk1[0].c
                |vpiIndex:
                \_constant: , line:16, parent:work@dut.genblk1[0].c
                  |vpiConstType:9
                  |vpiDecompile:0
                  |vpiSize:64
                  |UINT:0
              |vpiRhs:
              \_operation: , line:16, parent:work@dut.genblk1[0]
                |vpiOpType:33
                |vpiOperand:
                \_constant: , line:16
                  |vpiConstType:9
                  |vpiDecompile:10
                  |vpiSize:4
                  |UINT:10
```
`vpiDecompile` for the argument is just `10`, without base or size. I would expect here the original code: `4'b11`.
This causes issues in Verilator:
* in general, we leave size inference to later passes, as unsized literals are fine in e.g. bit/var selects, setting explicit size for each and every `vpiConstant` is not wanted
* we cannot explicitly set the size for specific operations  - operation arguments can potentially be more complicated
* we cannot set the size only for direct operands - they are missing parent handles, but that's likely a separate issue

I can see two potential fixes for this:
* do not substitute enum reference with the value, possibly with a switch similar to `noparametersubstitution` - similar to #1013 
* provide original values for the enum constants in the vpiDecompile field

Is any of those two feasible or is there a better approach?

alaindargelas, 2021-04-07:
I added the vpiDecompile and vpiSize properties to const_enum, that seemed simpler even if that does not follow the standard.
#1175
Please confimr that works.

rkapuscik, 2021-04-13:
I see that the enum constants now have the fields you mentioned, but the values where enums were substituted do not - `vpiDecompile` in `vpiRhs` is the same as in the initial comment (`3` vs `4b11` in enum) and there is no information that something was substituted to retrieve original value.

alaindargelas, 2021-04-14:
As soon as operators manipulate values, like concat the math only happen in one of the following base units: uint, int or double.
It is very complicated to keep the base units like hex, or binary while doing the math.
Now the size of the constant is propagated through the operations.

Can you explain what is the problem you are facing in Verilator now that you have the size of the constants?

alaindargelas, 2021-04-14:
Re-assign after providing in depth explanation about the Verilator issue.

rkapuscik, 2021-04-23:
I'll try to explain the problem sticking to the test case linked before: https://github.com/antmicro/uhdm-integration/blob/rk/update-ram2p/tests/EnumConcat/dut.v

Even though now the constant nodes have hardcoded sizes according to their types, we would need to reconstruct the constant in Verilator - either for each constant node or depending on context.

Adding sizes to each node bites us really quickly, for example in bit select nodes: when `c[i]` gets expanded, Verilator throws warnings:
```
%Warning-WIDTH: Bit extraction of array[3:0] requires 2 bit index, not 64 bits.
                        : ... In instance dut
-node: ARRAYSEL 0x5586e17f7a40 <e590#> {d0aa} @dt=0x5586e184ab60@(nw4)
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
```
The UHDM node looks as follows:
```
\_bit_select: (work@dut.genblk1[1].c), line:16:3, endln:16:4
  |vpiName:c
  |vpiFullName:work@dut.genblk1[1].c
  |vpiIndex:
  \_constant: , line:16:5, endln:16:6, parent:work@dut.genblk1[1].c
    |vpiConstType:9
    |vpiDecompile:1
    |vpiSize:64
    |UINT:1
```
While `vpiSize:64` is not incorrect, Verilator expects the size to be limited to the array size. In vanilla Verilator frontend, this size is first stored as "unset":
```
1:2:1:2:1:2: SELBIT 0x5646663bdc80 <e143> {c16af}
1:2:1:2:1:2:1: VARREF 0x5646663bae20 <e192> {c16ae} @dt=0@  c [LV] => VAR 0x564666348420 <e71> {c12ao} @dt=0@  c [VSTATIC]  VAR
1:2:1:2:1:2:2: CONST 0x5646663bb4e0 <e400#> {c14ac} @dt=0x5646663c8c30@(G/wu32/1)  ?32?h1
1:2:1:2:1:2:4: ATTROF 0x5646663bdd40 <e208#> {c16af} [VAR_BASE]
1:2:1:2:1:2:4:1: VARREF 0x5646663bb2a0 <e207> {c16ae} @dt=0@  c [RV] <- VAR 0x564666348420 <e71> {c12ao} @dt=0@  c [VSTATIC]  VAR
```
then adjusted to the signal size (after the width inference):
```
1:2:1:2:1:2: ARRAYSEL 0x5646663bef40 <e869#> {c16af} @dt=0x5646663c8750@(nw4)
1:2:1:2:1:2:1: VARREF 0x5646663bae20 <e865#> {c16ae} @dt=0x5646663cbcf0@(nw4)u[0:3]  c [LV] => VAR 0x564666348420 <e821#> {c12ao} @dt=0x5646663cbcf0@(nw4)u[0:3]  c [VSTATIC]  VAR
1:2:1:2:1:2:2: CONST 0x5646663f2640 <e885#> {c14ac} @dt=0x5646663f4000@(G/w2)  2'h1
```
Resolution of the VARREF happens at later stage and is not done in the UHDM frontend.

This behavior prevents us from adding size to each and every node, or even each node under replication operation since we can encounter a bit select or similar operation in the arguments.
Adding size only for direct operands of the concatenation/replication nodes is not possible as parent handles are not set for the vpiConstant nodes.

The two solutions would solve this in following ways:
* by having the original `vpiDecompile` value in the node substituted for enum reference, we can skip adding sizes manually and pass the value to Verilator to set correct size in later stages. This is how it was done up to now.
* by disabling enum substitutions we are getting even closer to original Verilator behavior. At the beginning original Verilator AST is:
```
1:2:2:4:2:2:1: ASSIGN 0x5646663bcc00 <e141#> {c16aj} @dt=0@
1:2:2:4:2:2:1:1: REPLICATE 0x5646663bcb40 <e142#> {c16al} @dt=0x5646663c8dd0@(G/w0)
1:2:2:4:2:2:1:1:1: PARSEREF 0x56466636aa80 <e135#> {c16am}  PRIV_LVL_M [TEXT]
1:2:2:4:2:2:1:1:2: CONST 0x5646663b90e0 <e136#> {c16al} @dt=0x5646663c8d00@(G/w32)  32'h1
1:2:2:4:2:2:1:2: SELBIT 0x5646663bca80 <e143#> {c16af}
1:2:2:4:2:2:1:2:1: PARSEREF 0x56466636a8c0 <e126#> {c16ae}  c [TEXT]
1:2:2:4:2:2:1:2:2: PARSEREF 0x56466636a9a0 <e127#> {c16ag}  i [TEXT]
```
(the `32'h1` const is added during replicate node construction)
Note the `PARSEREF PRIV_LVL_M` - it then gets resolved to the original enum, that in UHDM already would have the correct `vpiDecompile`. Having a similar switch would also possibly allow for workarounds similar to those used with parameters (see #1013), though I haven't encountered any for now.

I am open to alternative solutions if you have any suggestions.

As a note, I identified the commit that introduced the breaking change: https://github.com/alainmarcel/Surelog/commit/10812485a444ff808a95367321a4e92fbe41016d

alaindargelas, 2021-04-24:
#1226 should take care of it, please confirm and close the issue if resolved.

rkapuscik, 2021-04-27:
This test case works now, thank you. 
I see other related failures that were probably masked before, but I need to investigate them first and will report as new issues if relevant.

closed 2021-04-27 by rkapuscik