chipsalliance/Surelog #1185: Missing initial value for output reg

kamilrakoczy, 2021-04-09:
TEST:
```
module dut (b);                                                                                          
        wire a = 0;      
        output reg b = 0;                                                                        
        always @* begin                                                                                  
                if(a) b = 1;                                                                     
        end                                                                                              
endmodule
```
UHDM lacks of information about initial ``b`` value it only have information about ``a``:
```
|vpiContAssign:                                                                                       
   \_cont_assign: , line:2, parent:work@dut                                                              
     |vpiRhs:                                                                                                                                                                                                      
     \_constant: , line:2                                                                                                                                                                                          
       |vpiConstType:9                                                                                                                                                                                             
       |vpiDecompile:0                                                                                                                                                                                             
       |vpiSize:64                                                                                       
     |vpiLhs:                                                                                                                                                                                                      
     \_logic_net: (work@dut.a), line:2, parent:work@dut
```
Changing test to:
```
module dut (b);                                                                                          
        wire a = 0;      
        reg b = 0;                                                                        
        always @* begin                                                                                  
                if(a) b = 1;                                                                     
        end                                                                                              
endmodule
```
Provides initial ``b`` value:
```
   |vpiContAssign:                                                                                                                                                                                                 
   \_cont_assign: , line:2, parent:work@dut                                                                                                                                                                        
     |vpiRhs:                                                                                                                                                                                                      
     \_constant: , line:2                                                                                                                                                                                          
       |vpiConstType:9                                                                                                                                                                                             
       |vpiDecompile:0                                                                                                                                                                                             
       |vpiSize:64                                                                                                                                                                                                 
     |vpiLhs:                                                                                            
     \_logic_net: (work@dut.a), line:2, parent:work@dut                                                  
   |vpiContAssign:                                                                                                                                                                                                 
   \_cont_assign: , line:3, parent:work@dut                                                              
     |vpiRhs:                                                                                            
     \_constant: , line:3                                                                                
       |vpiConstType:9                                                                                                                                                                                             
       |vpiDecompile:0                                                                                                                                                                                             
       |vpiSize:64                                                                                                                                                                                                 
     |vpiLhs:                                                                                                                                                                                                      
     \_logic_net: (work@dut.b), line:3, parent:work@dut
```

kamilrakoczy, 2021-04-09:
Also I think, ``reg b = 0;`` should be reported as initial instead of ``vpiContAssign``. According to standard ``10.5 Variable declaration assignment (variable initialization)``:
```
Unlike nets, a variable cannot have an implicit continuous assignment as part of its declaration. An
assignment as part of the declaration of a variable is a variable initialization, not a continuous assignment.
The variable declaration assignment is a special case of procedural assignment as it assigns a value to a
variable. It allows an initial value to be placed in a variable in the same statement that declares the variable
(see 6.8). The assignment does not have duration; instead, the variable holds the value until the next
assignment to that variable.
For example:
wire w = vara & varb; // net with a continuous assignment
logic v = consta & constb; // variable with initialization
Setting the initial value of a static variable as part of the variable declaration (including static class
members) shall occur before any initial or always procedures are started. See also 6.21.
```

alaindargelas, 2021-04-10:
There is a property in cont_assign that I was not using: vpiNetDeclAssign.
With #1186 I do now.
This property along with the vpiNetType should allow you to make the distinction in between a real cont_assign and an inital value assign. Along with the net type, you can choose to act accordingly.

PR #1187 (2021-04-10): Default init port value
Fixes #1185
closed 2021-04-10 by alaindargelas