chipsalliance/Surelog #1219: Language Server

hs-apotell, 2021-04-20:
Is there a language server implementation that uses Surelog? How about a linter?

I found a few Verilog/SystemVerilog Language Servers but none of them are built on top of Surelog. Do you/anyone know of one in works or readily available for use?

Few other similar implementation based on various parsers -

https://github.com/dalance/svls
https://github.com/sthenic/vls
https://github.com/dalance/svlint
https://github.com/vivekmalneedi/veridian

mithro, 2021-04-20:
I would love to see this happen!

alaindargelas, 2021-04-21:
Not that I'm aware of. As @mithro said, that would be an awesome application.
Surelog is mainly used as a front-end for Yosys and Verilator (open-source), also I heard of some design-level analyzers, although I think they are not open-source (Linter for design properties like elaboration values, or state machine analysis).

hs-apotell, 2021-04-21:
I will spend some time investigating how the existing solutions go about doing it and see if I can put together a quick (and likely dirty) prototype built on top of Surelog.

alaindargelas, 2021-04-22:
Let me know if any changes needed

closed 2021-04-30 by alaindargelas