chipsalliance/Surelog #1249: segmentation faults when parsing Opentitan system Verilog top files

sarahSherifAli, 2021-05-04:
When parsing the system Verilog files in the Makefile.in 
[Makefile.in.txt](https://github.com/alainmarcel/Surelog/files/6419448/Makefile.in.txt)

 using this command line: 
-parse -sverilog -d coveruhdm ${SURELOG_FLAGS} $(INCLUDE) $(TOP_FILE).
 where
SURELOG_FLAGS := --disable-feature=parametersubstitution +define+SYNTHESIS
INCLUDE := -I$(EARLGRAY_BUILD)/src/lowrisc_prim_assert_0.1/rtl \
        -I$(EARLGRAY_BUILD)/src/lowrisc_prim_util_memload_0/rtl \

I'm running into segmentation faults. a screen shoot contains frame that throw the segmentation fault  
[image]

alaindargelas, 2021-05-04:
Since I do not know your Makefile system, I'll need the following:

- The actual full command line used for Surelog (check slpp_all/surelog.log)
   COMMAND:   < command line >

Also the actual source code for the Opentitan version you are using:

- A github link to the version of the source code for Opentitan you are using with instructions to build it.
  Or 
- a tar.gz file with only the source code (SystemVerilog) you are compiling.

sarahSherifAli, 2021-05-04:
- the tar.gz file contains the source codes, the  Makefile, and the surelog executable file

- COMMAND: gdb -args  < path to surelog executable >/surelog -parse -sverilog -d coveruhdm ${SURELOG_FLAGS} $(INCLUDE) $(TOP_FILE)

[seg_fault_titan.tar.gz](https://github.com/alainmarcel/Surelog/files/6420349/seg_fault_titan.tar.gz)

alaindargelas, 2021-05-04:
I still need the command line for Surelog fully expanded, no $(...) remaining.
Please copy the command line from the surelog.log file COMMAND: section  or just attach the surelog.log file to this request.

alaindargelas, 2021-05-05:
Please attach the surelog.log file to this issue. I'll get the complete command line without having to debug your Makefile.

alain@alain-i7:~/Surelog/third_party/tests/Titan/issue$ make 
Makefile:17: tests/opentitan/Makefile.in: No such file or directory
make: *** No rule to make target 'tests/opentitan/Makefile.in'.  Stop.
alain@alain-i7:~/Surelog/third_party/tests/Titan/issue$ find . -name Makefile.in
alain@alain-i7:~/Surelog/third_party/tests/Titan/issue$

sarahSherifAli, 2021-05-05:
[surelog.log](https://github.com/alainmarcel/Surelog/files/6425900/surelog.log)

PR #1255 (2021-05-05): fix core dump
Fixes #1249
closed 2021-05-05 by alaindargelas