chipsalliance/Surelog #1251: Parameter Assign Rhs() value assign As constant 'UNIT' in UHDM Output

bsp13, 2021-05-05:
If IDLE = 2'b00 than why it is IDLE = UINT:0?

Example:

```
module fsm_2_always_block_value (output reg rd, ds, input go, ws, clk, rst_n);

parameter IDLE = 2'b00, READ = 2'b01, DLY = 2'b11, DONE = 2'b10;  ---> assign  binary..
reg [1:0] state, next;

endmodule
```

UHDM Output:
```
 |vpiParamAssign:
  \_param_assign: , line:3:10, endln:3:22, parent:work@fsm_2_always_block_value
    |vpiRhs:
    \_constant: , line:3:17, endln:3:22
      |vpiConstType:9
      |vpiDecompile:0
      |vpiSize:2
      |UINT:0 <----------------------------------------UINT:0 but It is BIN:00
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.IDLE), line:3:10, endln:3:22, parent:work@fsm_2_always_block_value
      |vpiName:IDLE
      |vpiFullName:work@fsm_2_always_block_value.IDLE
      |UINT:0
  |vpiParamAssign:
  \_param_assign: , line:3:24, endln:3:36, parent:work@fsm_2_always_block_value
    |vpiRhs:
    \_constant: , line:3:31, endln:3:36
      |vpiConstType:9
      |vpiDecompile:1
      |vpiSize:2
      |UINT:1
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.READ), line:3:24, endln:3:36, parent:work@fsm_2_always_block_value
      |vpiName:READ
      |vpiFullName:work@fsm_2_always_block_value.READ
      |UINT:1
  |vpiParamAssign:
  \_param_assign: , line:3:38, endln:3:49, parent:work@fsm_2_always_block_value
    |vpiRhs:
    \_constant: , line:3:44, endln:3:49
      |vpiConstType:9
      |vpiDecompile:3
      |vpiSize:2
      |UINT:3
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.DLY), line:3:38, endln:3:49, parent:work@fsm_2_always_block_value
      |vpiName:DLY
      |vpiFullName:work@fsm_2_always_block_value.DLY
      |UINT:3
  |vpiParamAssign:
  \_param_assign: , line:3:51, endln:3:63, parent:work@fsm_2_always_block_value
    |vpiRhs:
    \_constant: , line:3:58, endln:3:63
      |vpiConstType:9
      |vpiDecompile:2
      |vpiSize:2
      |UINT:2
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.DONE), line:3:51, endln:3:63, parent:work@fsm_2_always_block_value
      |vpiName:DONE
      |vpiFullName:work@fsm_2_always_block_value.DONE
      |UINT:2
```

alaindargelas, 2021-05-05:
Constants are reduced to UINT, INT, REAL and STRINGS so math can be done more easily in the elaborated model.
Maintaining hex or bin or oct base is not desired in the elaborated model as it would make maths more complicated.
Note that the non-elaborated model keeps the constants as in the the source text:

```
module fsm_2_always_block_value ();

parameter IDLE = 2'b00, READ = 2'b01;  

endmodule

design: (work@fsm_2_always_block_value), id:12
|vpiName:work@fsm_2_always_block_value
|vpiElaborated:1
|uhdmallModules:         <<<<<<<<<<<   Non elaborated model
\_module: work@fsm_2_always_block_value (work@fsm_2_always_block_value), id:13 tests/ScratchPad.sv:1: , endln:5:9, parent:work@fsm_2_always_block_value, parID:12
  |vpiDefName:work@fsm_2_always_block_value
  |vpiFullName:work@fsm_2_always_block_value
  |vpiParamAssign:
  \_param_assign: , id:1, line:3:10, endln:3:22, parent:work@fsm_2_always_block_value, parID:13
    |vpiRhs:
    \_constant: , id:2, line:3:17, endln:3:22
      |vpiConstType:3
      |vpiDecompile:2'b00
      |vpiSize:2
      |BIN:00      <<<<< Original constant as in the text
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.IDLE), id:0, line:3:10, endln:3:22, parent:work@fsm_2_always_block_value, parID:13
      |vpiName:IDLE
      |vpiFullName:work@fsm_2_always_block_value.IDLE
  |vpiParamAssign:
  \_param_assign: , id:4, line:3:24, endln:3:36, parent:work@fsm_2_always_block_value, parID:13
    |vpiRhs:
    \_constant: , id:5, line:3:31, endln:3:36
      |vpiConstType:3
      |vpiDecompile:2'b01
      |vpiSize:2
      |BIN:01           <<<<< Original constant as in the text
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.READ), id:3, line:3:24, endln:3:36, parent:work@fsm_2_always_block_value, parID:13
      |vpiName:READ
      |vpiFullName:work@fsm_2_always_block_value.READ
  |vpiParameter:
  \_parameter: (work@fsm_2_always_block_value.IDLE), id:0, line:3:10, endln:3:22, parent:work@fsm_2_always_block_value, parID:13
  |vpiParameter:
  \_parameter: (work@fsm_2_always_block_value.READ), id:3, line:3:24, endln:3:36, parent:work@fsm_2_always_block_value, parID:13
|uhdmtopModules:        <<<<<<<<<<<  Elaborated model
\_module: work@fsm_2_always_block_value (work@fsm_2_always_block_value), id:14 tests/ScratchPad.sv:1: , endln:5:9
  |vpiDefName:work@fsm_2_always_block_value
  |vpiName:work@fsm_2_always_block_value
  |vpiParamAssign:
  \_param_assign: , id:8, line:3:10, endln:3:22, parent:work@fsm_2_always_block_value, parID:14
    |vpiRhs:
    \_constant: , id:9, line:3:17, endln:3:22
      |vpiConstType:9
      |vpiDecompile:0
      |vpiSize:2
      |UINT:0
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.IDLE), id:15, line:3:10, endln:3:22, parent:work@fsm_2_always_block_value, parID:14
      |vpiName:IDLE
      |vpiFullName:work@fsm_2_always_block_value.IDLE
      |UINT:0
  |vpiParamAssign:
  \_param_assign: , id:10, line:3:24, endln:3:36, parent:work@fsm_2_always_block_value, parID:14
    |vpiRhs:
    \_constant: , id:11, line:3:31, endln:3:36
      |vpiConstType:9
      |vpiDecompile:1
      |vpiSize:2
      |UINT:1
    |vpiLhs:
    \_parameter: (work@fsm_2_always_block_value.READ), id:16, line:3:24, endln:3:36, parent:work@fsm_2_always_block_value, parID:14
      |vpiName:READ
      |vpiFullName:work@fsm_2_always_block_value.READ
      |UINT:1
  |vpiParameter:
  \_parameter: (work@fsm_2_always_block_value.IDLE), id:15, line:3:10, endln:3:22, parent:work@fsm_2_always_block_value, parID:14
  |vpiParameter:
  \_parameter: (work@fsm_2_always_block_value.READ), id:16, line:3:24, endln:3:36, parent:work@fsm_2_always_block_value, parID:14

```

hs-apotell, 2021-05-05:
> Constants are reduced to UINT, INT, REAL and STRINGS so math can be done more easily in the elaborated model.

This change would make the models non-deterministic. The location information for the model wouldn't match the value. I don't think optimizations like these should be part of elaboration process.

alaindargelas, 2021-05-06:
Parameters are the main mechanism used in Elaboration to compute sizes. Math operations are commonly performed to compute the next set of parameters, so optimization like these are definitely part of the elaboration process.
On the other hand, the way the parameters are used in this example is not a recommended use. Enum const are preferred over parameters to describe state encoding. May I suggest you try using enums instead of parameters for state machine description?

hs-apotell, 2021-05-06:
> ... the way the parameters are used in this example ...

My argument isn't about this specific example. I was suggesting it for any consumer input, in general.

For a specific use case - we are using Surelog's elaborated model to generate FSM.  The view is interactive and can jump from FSM view to source code and back. Accurate location information becomes critical for this to work. For each state variable node, we also show the variable's constant value, if applicable and available. User will expect this presentation to match what the source reads.

Moreover, location being wrong is just one aspect. The width of the variable is also ambiguous with this conversion i.e. `b'00` would be `u0` and if variable was incremented every tick, after 4 ticks would be `b'00` again if treated as binary but as integer range is different. This adds unnecessary burden on users of Surelog to account for this on their end.

On a parallel note - Is there a `O(1)` method/approach to jump between elaborate and non-elaborated object i.e. without employing any search in the tree? Does that even make logical sense?

alaindargelas, 2021-05-06:
Fixed by #1263.
The compromise is that the value used at definition time is held by the param object, where the actual post-elaboration value is held by the param_assign.

There is no O(1) approach to navigate in between the elaborated and non-elaborated tree.

closed 2021-05-06 by alaindargelas