chipsalliance/Surelog #1252: Actual Group of local variable in function did not found

bsp13, 2021-05-05:
Verilog Example::
```
module vend( coin, clock, reset, newspaper);
//Input output port declarations
input [1:0] coin;
input clock;
input reset;
output newspaper;
wire newspaper;
//internal FSM state declarations
wire [1:0] NEXT_STATE;
reg [1:0] PRES_STATE;
//state encodings
parameter s0 = 2'b00;
parameter s5 = 2'b01;
parameter s10 = 2'b10;
parameter s15 = 2'b11;

//Combinational logic
function [2:0] fsm;
input [1:0] fsm_coin;
input [1:0] fsm_PRES_STATE;
reg fsm_newspaper;
reg [1:0] fsm_NEXT_STATE;

begin
    case (fsm_PRES_STATE)   
    s0: //state = s0
    begin
        if (fsm_coin == 2'b10)
            begin
                fsm_newspaper = 1'b0;  <---------------------For this assignment
                fsm_NEXT_STATE = s10;  <---------------------For this assignment
            end
        else if (fsm_coin == 2'b01)
            begin
                fsm_newspaper = 1'b0;   <---------------------For this assignment
                fsm_NEXT_STATE = s5;    <---------------------For this assignment
            end
        else
            begin
                fsm_newspaper = 1'b0;  <---------------------For this assignment
                fsm_NEXT_STATE = s0;   <---------------------For this assignment
            end
        end   
    endcase
fsm = {fsm_newspaper, fsm_NEXT_STATE};
end
endfunction

always @(posedge clock)
begin
  assign {newspaper, NEXT_STATE} = fsm(coin, PRES_STATE);
end

always @(posedge clock)
    begin
        if (reset == 1'b1)
            PRES_STATE <= s0;
        else
            PRES_STATE <= NEXT_STATE;
        end
endmodule
```

UHDM Output:

```
  |vpiLhs:
          \_ref_obj: (work@vend.fsm.fsm_newspaper), line:32:16, endln:32:29, parent:work@vend.fsm
         |vpiName:fsm_newspaper
         |vpiFullName:work@vend.fsm.fsm_newspaper

 |vpiLhs:
         \_ref_obj: (work@vend.fsm.fsm_NEXT_STATE), line:33:16, endln:33:30, parent:work@vend.fsm
         vpiName:fsm_NEXT_STATE
         vpiFullName:work@vend.fsm.fsm_NEXT_STATE
         
|vpiLhs:
         \_ref_obj: (work@vend.fsm.fsm_newspaper), line:37:16, endln:37:29, parent:work@vend.fsm
         |vpiName:fsm_newspaper
         |vpiFullName:work@vend.fsm.fsm_newspaper

|vpiLhs:
         \_ref_obj: (work@vend.fsm.fsm_NEXT_STATE), line:43:16, endln:43:30, parent:work@vend.fsm
         |vpiName:fsm_NEXT_STATE
        |vpiFullName:work@vend.fsm.fsm_NEXT_STATE
```

alaindargelas, 2021-05-05:
I think you are not traversing the function definition in the elaborated model properly,
The binding to the actual objects is happening. Make sure the UHDM Elaboration is on, and you are traversing the uhdmtopModules relation, then you are traversing the vpiTaskFunc relation.
See below actuals:
```
|uhdmtopModules:
...
 |vpiTaskFunc:
  \_function: (work@vend.fsm), id:185, line:18, endln:47:11, parent:work@vend, parID:156
      .....
               |vpiElseStmt:
                \_begin: (work@vend.fsm), id:215, line:39:12, endln:42:15, parID:206
                  |vpiFullName:work@vend.fsm
                  |vpiStmt:
                  \_assignment: , id:216, line:40:16, endln:40:36, parent:work@vend.fsm, parID:215
                    |vpiOpType:82
                    |vpiBlocking:1
                    |vpiLhs:
                    \_ref_obj: (work@vend.fsm.fsm_newspaper), id:217, line:40:16, endln:40:29, parID:216
                      |vpiName:fsm_newspaper
                      |vpiFullName:work@vend.fsm.fsm_newspaper
                      |vpiActual:
                      \_logic_var: (work@vend.fsm.fsm_newspaper), id:188, line:21, endln:21:3, parent:work@vend.fsm, parID:185
                    |vpiRhs:
                    \_constant: , id:69, line:40:32, endln:40:36, parID:70
                  |vpiStmt:
                  \_assignment: , id:218, line:41:16, endln:41:35, parent:work@vend.fsm, parID:215
                    |vpiOpType:82
                    |vpiBlocking:1
                    |vpiLhs:
                    \_ref_obj: (work@vend.fsm.fsm_NEXT_STATE), id:219, line:41:16, endln:41:30, parID:218
                      |vpiName:fsm_NEXT_STATE
                      |vpiFullName:work@vend.fsm.fsm_NEXT_STATE
                      |vpiActual:
                      \_logic_var: (work@vend.fsm.fsm_NEXT_STATE), id:189, line:22, endln:22:3, parent:work@vend.fsm, parID:185
```

Do get the function stmts from the function definition, do not traverse the vpiInstance relation which takes you to a somewhat recursive traversal.
The actual binding only occurs at the first levels of traversals, not when you are taken back to a definition by relations like vpiInstance (To avoid memory blowups).

bsp13, 2021-05-06:
I am using below code for traversing function definition.  is it same pointer for function to listen elaborated function definition?

```C++
virtual void enterFunc_call(
    const UHDM::func_call *object, const UHDM::BaseClass *parent,
    vpiHandle handle, vpiHandle parentHandle) override
  {
     UHDM::listen_any(NewVpiHandle(object->Function()), this);
  }

virtual void enterTask_call(
    const UHDM::task_call *object, const UHDM::BaseClass *parent,
    vpiHandle handle, vpiHandle parentHandle) override
  {
   UHDM::listen_any(NewVpiHandle(object->Task()), this);
  }
```

alaindargelas, 2021-05-06:
I see. That is the problem. When you are traversing from the function call, it is not the same pointer to the function definition as when you traverse the function definition from the module instance (That one has the actuals filled). 
The pointer is not the same because the elaborator does not do hash consing for functions, remember the elaborator is auto generated code that is pretty brute force. It is creating a new copy of the function definition. 

I can try to do function definition hash consing. But I'm afraid that is going to be bogus for parameterized function those arguments have width depending on parameters, which needs to be uniquified.
As of now the elaborator is pretty dumb, it uniquifies everything excepts constants and typespecs.

Can you get the function definition from the module? You'll get the vpiActual for all the variables.
Then, when you traverse the function call, simply get the function definition name and "bind" it to the function definition you collected first.

hs-apotell, 2021-05-08:
That worked. Thanks!

closed 2021-05-08 by hs-apotell

alaindargelas, 2021-05-09:
Good to hear, I also fixed the Instance relation in #1276, it was actually pointing to the non-elaborated tree.
This is much cleaner, the elaborated tree should not have pointers to the non-elaborated tree.