chipsalliance/Surelog #1282: Pass by value vs. Pass by reference

hs-apotell, 2021-05-11:
How to differentiate between function arguments passed by value vs. reference?

I see `UHDM::io_decls` have the `VpiDirection` member but can't find any reference to what each constant means. I see a value of 6 when using 'ref' and a value of 1 when I don't. Are there defines/enums/constants defined for what these constants mean?

alaindargelas, 2021-05-11:
Please include a small testcase

alaindargelas, 2021-05-13:
From: 
https://github.com/alainmarcel/UHDM/blob/3f829bf73fe2cdc9827746c68c2b8df6f6362a03/include/sv_vpi_user.h#L279
#define vpiRef                             6 

When decoding VPI types, check both the vpi_user.h (Verilog) and sv_vpi_user.h file which extends Verilog with SystemVerilog types.

closed 2021-05-13 by alaindargelas