chipsalliance/Surelog #1306: flatbuffer broke Surelog build

alaindargelas, 2021-05-18:
@tgorochowik, can you help set the flatbuffer subsystem to a couple of days ago version? The latest version is broken for me.
Can you describe here the steps you took to fix the capnp version last time? I can't reproduce it for flatbuffer.

https://github.com/google/flatbuffers/pull/6620

tgorochowik, 2021-05-19:
@alaindargelas the latest submodule update does not include the PR you mention. The CI fails because of some code formatting issues, the PR you point to is not used in Surelog at the moment (at least not in the repo, make sure you did not update the submodule locally)

But in general, when you want to change the revision of a submodule remember it is just another git repository, so you can for example:

```
cd third_party/flatbuffers
git checkout <whatever revision works for you>
cd ../..
git add third_party/flatbuffers
git commit -m "Update flatbuffers rev" 
```

alaindargelas, 2021-05-20:
Thanks @tgorochowik. 
The flatbuffer guys reverted their change, I'll provide them with my sandbox experiment.
I got it now, I was the only one seeing this issue since I did not commit it.

closed 2021-05-20 by alaindargelas