chipsalliance/Surelog #1314: Bit select on struct members dropped

kbieganski, 2021-05-20:
Verilog code:
```systemverilog
module t;
  typedef struct packed {
    logic        q;
  } rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t;

  typedef struct packed {
    rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t [0:0] sw_rst_ctrl_n;
  } rstmgr_reg2hw_t;

  rstmgr_reg2hw_t reg2hw;
  rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t [0:0] sw_rst_ctrl_n;

  logic X = reg2hw.sw_rst_ctrl_n[0].q;
  logic Y = sw_rst_ctrl_n[0].q;
endmodule;
```
UHDM (part of it):
```
  |vpiContAssign:
  \_cont_assign: , line:13:8, endln:13:9, parent:work@t
    |vpiNetDeclAssign:1
    |vpiRhs:
    \_hier_path: (sw_rst_ctrl_n[0].q), line:13:12, endln:13:30
      |vpiName:sw_rst_ctrl_n[0].q
      |vpiActual:
      \_bit_select: (sw_rst_ctrl_n)
        |vpiName:sw_rst_ctrl_n
        |vpiIndex:
        \_constant: , line:13:26, endln:13:27
          |vpiConstType:9
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
      |vpiActual:
      \_ref_obj: (q)
        |vpiName:q
    |vpiLhs:
    \_logic_net: (work@t.X), line:13:8, endln:13:9, parent:work@t
      |vpiName:X
      |vpiFullName:work@t.X
      |vpiNetType:36
  |vpiContAssign:
  \_cont_assign: , line:14:8, endln:14:9, parent:work@t
    |vpiNetDeclAssign:1
    |vpiRhs:
    \_hier_path: (reg2hw.sw_rst_ctrl_n.q), line:14:12, endln:14:37
      |vpiName:reg2hw.sw_rst_ctrl_n.q
      |vpiActual:
      \_ref_obj: (reg2hw)
        |vpiName:reg2hw
      |vpiActual:
      \_ref_obj: (sw_rst_ctrl_n)
        |vpiName:sw_rst_ctrl_n
      |vpiActual:
      \_ref_obj: (q)
        |vpiName:q
    |vpiLhs:
    \_logic_net: (work@t.Y), line:14:8, endln:14:9, parent:work@t
      |vpiName:Y
      |vpiFullName:work@t.Y
      |vpiNetType:36
```
The first `cont_assign` has a bit select on the RHS (which is correct), but it's dropped for the second one.

Not sure if it behaves differently for packed/unpacked array/struct types.

alaindargelas, 2021-05-21:
I'm not seeing the same output. Check the elaborated section below.
Do you have an old version?

```
====== UHDM =======
design: (work@t), id:31
|vpiName:work@t
|vpiElaborated:1
|uhdmallModules:
\_module: work@t (work@t), id:32 tests/ScratchPad.sv:1: , endln:18:9, parent:work@t, parID:31
  |vpiDefName:work@t
  |vpiFullName:work@t
  |vpiNet:
  \_logic_net: (work@t.reg2hw), id:33, line:11:18, parent:work@t, parID:32
    |vpiName:reg2hw
    |vpiFullName:work@t.reg2hw
  |vpiNet:
  \_logic_net: (work@t.sw_rst_ctrl_n), id:34, line:13:43, parent:work@t, parID:32
    |vpiName:sw_rst_ctrl_n
    |vpiFullName:work@t.sw_rst_ctrl_n
  |vpiNet:
  \_logic_net: (work@t.X), id:35, line:15:7, parent:work@t, parID:32
    |vpiName:X
    |vpiFullName:work@t.X
    |vpiNetType:36
  |vpiNet:
  \_logic_net: (work@t.Y), id:36, line:17:8, parent:work@t, parID:32
    |vpiName:Y
    |vpiFullName:work@t.Y
    |vpiNetType:36
  |vpiTypedef:
  \_struct_typespec: (rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t), id:0, line:3:10, endln:3:16
    |vpiPacked:1
    |vpiName:rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t
    |vpiTypespecMember:
    \_typespec_member: (q), id:2, line:4:17, endln:4:18
      |vpiName:q
      |vpiTypespec:
      \_logic_typespec: , id:1, line:4:4, endln:4:9
  |vpiTypedef:
  \_struct_typespec: (rstmgr_reg2hw_t), id:3, line:7:10, endln:7:16
    |vpiPacked:1
    |vpiName:rstmgr_reg2hw_t
    |vpiTypespecMember:
    \_typespec_member: (sw_rst_ctrl_n), id:8, line:8:45, endln:8:58
      |vpiName:sw_rst_ctrl_n
      |vpiTypespec:
      \_packed_array_typespec: , id:7
        |vpiRange:
        \_range: , id:4, line:8:40, endln:8:43, parent:rstmgr_reg2hw_t, parID:3
          |vpiLeftRange:
          \_constant: , id:5, line:8:40, endln:8:41, parID:4
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
          |vpiRightRange:
          \_constant: , id:6, line:8:42, endln:8:43, parID:4
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
        |vpiElemTypespec:
        \_struct_typespec: (rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t), id:0, line:3:10, endln:3:16
|uhdmtopModules:
\_module: work@t (work@t), id:37 tests/ScratchPad.sv:1: , endln:18:9
  |vpiDefName:work@t
  |vpiName:work@t
  |vpiContAssign:
  \_cont_assign: , id:38, line:15:7, endln:15:8, parent:work@t, parID:37
[... 93 lines cut ...]
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
      |vpiActual:
      \_ref_obj: (q), id:46, parent:sw_rst_ctrl_n[0].q, parID:44
        |vpiName:q
    |vpiLhs:
    \_logic_net: (work@t.Y), id:29, line:17:8, endln:17:9, parID:43
      |vpiName:Y
      |vpiFullName:work@t.Y
      |vpiNetType:36
  |vpiNet:
  \_logic_net: (work@t.X), id:23, line:15:7, endln:15:8, parID:38
  |vpiNet:
  \_logic_net: (work@t.Y), id:29, line:17:8, endln:17:9, parID:43
  |vpiVariables:
  \_struct_var: (work@t.reg2hw), id:9, line:11:18, endln:11:24, parent:work@t, parID:37
  |vpiVariables:
  \_packed_array_var: (work@t.sw_rst_ctrl_n), id:18, line:13:43, endln:13:56, parent:work@t, parID:37
  |vpiTypedef:
  \_struct_typespec: (rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t), id:0, line:3:10, endln:3:16
  |vpiTypedef:
  \_struct_typespec: (rstmgr_reg2hw_t), id:3, line:7:10, endln:7:16
===================
```

kbieganski, 2021-05-21:
My mistake, the UHDM I attached doesn't exactly correspond to the test. But the problem is still there, in your output. The bit select on this cont assign's RHS is missing (there's only a 3-element hierpath):

```
  |vpiContAssign:
  \_cont_assign: , id:38, line:15:7, endln:15:8, parent:work@t, parID:37
    |vpiNetDeclAssign:1
    |vpiRhs:
    \_hier_path: (reg2hw.sw_rst_ctrl_n.q), id:39, line:15:11, endln:15:36, parID:38
      |vpiName:reg2hw.sw_rst_ctrl_n.q
      |vpiActual:
      \_ref_obj: (reg2hw), id:40, parent:reg2hw.sw_rst_ctrl_n.q, parID:39
        |vpiName:reg2hw
        |vpiActual:
        \_struct_var: (work@t.reg2hw), id:9, line:11:18, endln:11:24, parent:work@t, parID:37
          |vpiName:reg2hw
          |vpiFullName:work@t.reg2hw
          |vpiAutomatic:1
          |vpiVisibility:1
          |vpiTypespec:
          \_struct_typespec: (rstmgr_reg2hw_t), id:3, line:7:10, endln:7:16
            |vpiPacked:1
            |vpiName:rstmgr_reg2hw_t
            |vpiTypespecMember:
            \_typespec_member: (sw_rst_ctrl_n), id:8, line:8:45, endln:8:58
              |vpiName:sw_rst_ctrl_n
              |vpiTypespec:
              \_packed_array_typespec: , id:7
                |vpiRange:
                \_range: , id:4, line:8:40, endln:8:43, parent:rstmgr_reg2hw_t, parID:3
                  |vpiLeftRange:
                  \_constant: , id:5, line:8:40, endln:8:41, parID:4
                    |vpiConstType:9
                    |vpiDecompile:0
                    |vpiSize:64
                    |UINT:0
                  |vpiRightRange:
                  \_constant: , id:6, line:8:42, endln:8:43, parID:4
                    |vpiConstType:9
                    |vpiDecompile:0
                    |vpiSize:64
                    |UINT:0
                |vpiElemTypespec:
                \_struct_typespec: (rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t), id:0, line:3:10, endln:3:16
                  |vpiPacked:1
                  |vpiName:rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t
                  |vpiTypespecMember:
                  \_typespec_member: (q), id:2, line:4:17, endln:4:18
                    |vpiName:q
                    |vpiTypespec:
                    \_logic_typespec: , id:1, line:4:4, endln:4:9
      |vpiActual:
      \_ref_obj: (sw_rst_ctrl_n), id:41, parent:reg2hw.sw_rst_ctrl_n.q, parID:39
        |vpiName:sw_rst_ctrl_n
        |vpiActual:
        \_packed_array_var: (work@t.sw_rst_ctrl_n), id:18, line:13:43, endln:13:56, parent:work@t, parID:37
          |vpiName:sw_rst_ctrl_n
          |vpiFullName:work@t.sw_rst_ctrl_n
          |vpiAutomatic:1
          |vpiVisibility:1
          |vpiRange:
          \_range: , id:14, line:13:38, endln:13:41
            |vpiLeftRange:
            \_constant: , id:15, line:13:38, endln:13:39, parID:14
              |vpiConstType:9
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
            |vpiRightRange:
            \_constant: , id:16, line:13:40, endln:13:41, parID:14
              |vpiConstType:9
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
          |vpiElement:
          \_struct_var: (work@t.sw_rst_ctrl_n), id:17, parent:work@t.sw_rst_ctrl_n, parID:18
            |vpiFullName:work@t.sw_rst_ctrl_n
            |vpiTypespec:
            \_struct_typespec: (rstmgr_reg2hw_sw_rst_ctrl_n_mreg_t), id:0, line:3:10, endln:3:16
      |vpiActual:
      \_ref_obj: (q), id:42, parent:reg2hw.sw_rst_ctrl_n.q, parID:39
        |vpiName:q
    |vpiLhs:
    \_logic_net: (work@t.X), id:23, line:15:7, endln:15:8, parID:38
      |vpiName:X
      |vpiFullName:work@t.X
      |vpiNetType:36
```

PR #1315 (2021-05-22): select in hierechy path
Fixes #1314
closed 2021-05-22 by alaindargelas