chipsalliance/Surelog #1373: Windows Actions broken after move to ChipAlliance

alaindargelas, 2021-06-11:
@hs-apotell @kgugala @hzeller can you help?
After moving Surelog/UHDM to ChipAlliance, both Windows actions are broken, something to do with swig version.

hs-apotell, 2021-06-14:
Looks like the windows image now have swig pre-installed. I couldn't find anything in the notes [here](https://github.com/actions/virtual-environments/blob/main/images/win/Windows2019-Readme.md) but again it could be a unintended dependency of something else. Either way, the installed version is newer, v4.0 and AFAIK Surelog doesn't work with this version. So, we have three alternatives -
* I could try try installing the required version side-by-side and see if that works
* Try downgrading it but that might just break something else
* Update Surelog to work with newer version. That would have to come from @alaindargelas 

Any preferences? In the meantime I am going to attempt the side-by-side installation in a scratch repo.

alaindargelas, 2021-06-14:
I'll check later the problem with the latest version of swig

hs-apotell, 2021-06-14:
Side-by-side installation worked as a stop-gap solution until we could safely upgrade to using a newer version of swig.

closed 2021-06-14 by hs-apotell