chipsalliance/Surelog #1382: Bind error in assignment to function as return value

hs-apotell, 2021-06-15:
In example below, the LHS in the assignment operator resolves to null.

```
module example(clock);
  input clock;

  function int increment;
    input int a;
    input int b;
    begin
      increment = a + b;  // LHS RefObj resolves to null here ....
    end
  endfunction

  always @(posedge clock)
  begin
    x = increment(x, 1);
  end

endmodule
```

Here's the UHDM dump -
```
====== UHDM =======
design: (work@example)
|vpiName:work@example
|vpiElaborated:1
|uhdmallModules:
\_module: work@example (work@example) E:/Davenche/fsm/TestModules/example.v:1:1: , endln:17:10, parent:work@example
  |vpiDefName:work@example
  |vpiFullName:work@example
  |vpiProcess:
  \_always: , line:12:3, endln:15:6, parent:work@example
    |vpiAlwaysType:1
    |vpiStmt:
    \_event_control: , line:12:10, endln:15:6
      |vpiCondition:
      \_operation: , line:12:12, endln:12:19
        |vpiOpType:39
        |vpiOperand:
        \_ref_obj: (work@example.clock), line:12:20, endln:12:25
          |vpiName:clock
          |vpiFullName:work@example.clock
          |vpiActual:
          \_logic_net: (work@example.clock), line:1:16, endln:1:21, parent:work@example
            |vpiName:clock
            |vpiFullName:work@example.clock
      |vpiStmt:
      \_begin: (work@example), line:13:3, endln:15:6
        |vpiFullName:work@example
        |vpiStmt:
        \_assignment: , line:14:5, endln:14:24, parent:work@example
          |vpiOpType:82
          |vpiBlocking:1
          |vpiLhs:
          \_ref_obj: (work@example.x), line:14:5, endln:14:6, parent:work@example
            |vpiName:x
            |vpiFullName:work@example.x
          |vpiRhs:
          \_func_call: (increment), line:14:9, endln:14:24
            |vpiName:increment
            |vpiFunction:
            \_function: (work@example.increment), line:4:3, endln:10:14, parent:work@example
            |vpiArgument:
            \_ref_obj: (work@example.x), line:14:19, endln:14:20, parent:increment
              |vpiName:x
              |vpiFullName:work@example.x
            |vpiArgument:
            \_constant: , line:14:22, endln:14:23
              |vpiConstType:9
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
  |vpiPort:
  \_port: (clock), line:1:16, parent:work@example
    |vpiName:clock
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj:
      |vpiActual:
      \_logic_net: (work@example.clock), line:1:16, parent:work@example
        |vpiName:clock
        |vpiFullName:work@example.clock
  |vpiTaskFunc:
  \_function: (work@example.increment), line:4:3, endln:10:14, parent:work@example
    |vpiVisibility:1
    |vpiName:increment
    |vpiFullName:work@example.increment
    |vpiReturn:
    \_int_var: , line:4:12, endln:4:15
    |vpiInstance:
    \_module: work@example (work@example) E:/Davenche/fsm/TestModules/example.v:1:1: , endln:17:10, parent:work@example
    |vpiIODecl:
    \_io_decl: (a), line:5:15, endln:5:16, parent:work@example.increment
      |vpiName:a
      |vpiDirection:1
      |vpiTypedef:
      \_int_typespec: , line:5:11, endln:5:14
    |vpiIODecl:
    \_io_decl: (b), line:6:15, endln:6:16, parent:work@example.increment
      |vpiName:b
      |vpiDirection:1
      |vpiTypedef:
      \_int_typespec: , line:6:11, endln:6:14
    |vpiStmt:
    \_begin: (work@example.increment), line:7:5, endln:9:8, parent:work@example.increment
      |vpiFullName:work@example.increment
      |vpiStmt:
      \_assignment: , line:8:7, endln:8:24, parent:work@example.increment
        |vpiOpType:82
        |vpiBlocking:1
        |vpiLhs:
        \_ref_obj: (work@example.increment.increment), line:8:7, endln:8:16, parent:work@example.increment
          |vpiName:increment
          |vpiFullName:work@example.increment.increment
        |vpiRhs:
        \_operation: , line:8:19, endln:8:20, parent:work@example.increment
          |vpiOpType:24
          |vpiOperand:
          \_ref_obj: (work@example.increment.a), line:8:19, endln:8:20, parent:work@example.increment
            |vpiName:a
            |vpiFullName:work@example.increment.a
            |vpiActual:
            \_io_decl: (a), line:5:15, endln:5:16, parent:work@example.increment
          |vpiOperand:
          \_ref_obj: (work@example.increment.b), line:8:23, endln:8:24
            |vpiName:b
            |vpiFullName:work@example.increment.b
            |vpiActual:
            \_io_decl: (b), line:6:15, endln:6:16, parent:work@example.increment
  |vpiNet:
  \_logic_net: (work@example.clock), line:1:16, parent:work@example
|uhdmtopModules:
\_module: work@example (work@example) E:/Davenche/fsm/TestModules/example.v:1:1: , endln:17:10
  |vpiDefName:work@example
  |vpiName:work@example
  |vpiProcess:
  \_always: , line:12:3, endln:15:6, parent:work@example
    |vpiAlwaysType:1
    |vpiStmt:
    \_event_control: , line:12:10, endln:15:6
      |vpiCondition:
      \_operation: , line:12:12, endln:12:19
        |vpiOpType:39
        |vpiOperand:
        \_ref_obj: (work@example.clock), line:12:20, endln:12:25
          |vpiName:clock
          |vpiFullName:work@example.clock
          |vpiActual:
          \_logic_net: (work@example.clock), line:1:16, endln:1:21, parent:work@example
            |vpiName:clock
            |vpiFullName:work@example.clock
      |vpiStmt:
      \_begin: (work@example), line:13:3, endln:15:6
        |vpiFullName:work@example
        |vpiStmt:
        \_assignment: , line:14:5, endln:14:24, parent:work@example
          |vpiOpType:82
          |vpiBlocking:1
          |vpiLhs:
          \_ref_obj: (work@example.x), line:14:5, endln:14:6
            |vpiName:x
            |vpiFullName:work@example.x
          |vpiRhs:
          \_func_call: (increment), line:14:9, endln:14:24
            |vpiName:increment
            |vpiFunction:
            \_function: (work@example.increment), line:4:3, endln:10:14, parent:work@example
            |vpiArgument:
            \_ref_obj: (work@example.x), line:14:19, endln:14:20, parent:increment
              |vpiName:x
              |vpiFullName:work@example.x
            |vpiArgument:
            \_constant: , line:14:22, endln:14:23
              |vpiConstType:9
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
  |vpiPort:
  \_port: (clock), line:1:16, endln:1:21, parent:work@example
    |vpiName:clock
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj:
      |vpiActual:
      \_logic_net: (work@example.clock), line:1:16, endln:1:21, parent:work@example
  |vpiTaskFunc:
  \_function: (work@example.increment), line:4:3, endln:10:14, parent:work@example
    |vpiVisibility:1
    |vpiName:increment
    |vpiFullName:work@example.increment
    |vpiReturn:
    \_int_var: , line:4:12, endln:4:15
    |vpiInstance:
    \_module: work@example (work@example) E:/Davenche/fsm/TestModules/example.v:1:1: , endln:17:10
    |vpiIODecl:
    \_io_decl: (a), line:5:15, endln:5:16, parent:work@example.increment
      |vpiName:a
      |vpiDirection:1
      |vpiTypedef:
      \_int_typespec: , line:5:11, endln:5:14
    |vpiIODecl:
    \_io_decl: (b), line:6:15, endln:6:16, parent:work@example.increment
      |vpiName:b
      |vpiDirection:1
      |vpiTypedef:
      \_int_typespec: , line:6:11, endln:6:14
    |vpiStmt:
    \_begin: (work@example.increment), line:7:5, endln:9:8, parent:work@example.increment
      |vpiFullName:work@example.increment
      |vpiStmt:
      \_assignment: , line:8:7, endln:8:24, parent:work@example.increment
        |vpiOpType:82
        |vpiBlocking:1
        |vpiLhs:
        \_ref_obj: (work@example.increment.increment), line:8:7, endln:8:16
          |vpiName:increment
          |vpiFullName:work@example.increment.increment
        |vpiRhs:
        \_operation: , line:8:19, endln:8:20
          |vpiOpType:24
          |vpiOperand:
          \_ref_obj: (work@example.increment.a), line:8:19, endln:8:20
            |vpiName:a
            |vpiFullName:work@example.increment.a
            |vpiActual:
            \_io_decl: (a), line:5:15, endln:5:16, parent:work@example.increment
          |vpiOperand:
          \_ref_obj: (work@example.increment.b), line:8:23, endln:8:24
            |vpiName:b
            |vpiFullName:work@example.increment.b
            |vpiActual:
            \_io_decl: (b), line:6:15, endln:6:16, parent:work@example.increment
  |vpiNet:
  \_logic_net: (work@example.clock), line:1:16, endln:1:21, parent:work@example
===================
```

hs-apotell, 2021-06-21:
@alaindargelas After your change, LHS RefObj resolves to a logic_var. How to resolve the statement as a "return" statement rather than a regular logic variable assignment statement?

alaindargelas, 2021-06-22:
Actually Surelog is doing the right thing:

```
\_begin: (work@example.increment), id:43, line:7:5, endln:9:8, parent:work@example.increment, parID:40
      |vpiFullName:work@example.increment
      |vpiStmt:
      \_assignment: , id:44, line:8:7, endln:8:24, parent:work@example.increment, parID:43
        |vpiOpType:82
        |vpiBlocking:1
        |vpiLhs:
        \_ref_obj: (work@example.increment.increment), id:45, line:8:7, endln:8:16, parID:44
          |vpiName:increment
          |vpiFullName:work@example.increment.increment
          |vpiActual:
          \_int_var: , id:1, line:4:12, endln:4:15
```
- If does bind to a variable that matches the type of the return of function (int_var) not "logic_var" as you stated. 
- Surelog follows the LRM, in this case there was no return statement, so the implicit variable is enough.
- If the function uses a return statement, then Surelog will model it.

```
13.4.1 Return values and void functions
The function definition shall implicitly declare a variable, internal to the function, with the same name as the
function. This variable has the same type as the function return value. Function return values can be
specified in two ways, either by using a return statement or by assigning a value to the internal variable
with the same name as the function. For example:
function [15:0] myfunc1 (input [7:0] x,y);
myfunc1 = x * y - 1; // return value assigned to function name
endfunction
function [15:0] myfunc2 (input [7:0] x,y);
return x * y - 1; //return value is specified using return statement
endfunction
```

PR #1390 (2021-06-17): Log update for func return val binding
Fixes #1382
closed 2021-06-22 by alaindargelas