chipsalliance/Surelog #1385: Actual Group bind problem in vpiPrimitive:

bsp13, 2021-06-16:
System Verilog :
```
module dff_from_nand();
wire Q,Q_BAR;
reg D,CLK;

nand U1 (X,D,CLK) ;  <--------Actual group of X, D, CLK 

endmodule
```
Here, Actual group of X is nullptr.

alaindargelas, 2021-06-17:
Fixed by #1392

bsp13, 2021-06-21:
VpiParent bind problem in vpiPrimitive.Currently It is nullptr.
[image]

alaindargelas, 2021-06-21:
@bsp13, please attach a uhdm dump, not a screen snapshot and point to the field that is missing. Also is it the same testcase?

bsp13, 2021-06-21:
Surelog Output:
```
module dff_from_nand();
wire Q,Q_BAR;
reg D,CLK;

nand U1 (X,D,CLK) ;

endmodule
```

UHDM OUTPUT::
```
|uhdmtopModules:
\_module: work@dff_from_nand (work@dff_from_nand) C:\BHAVESH\Apotell\davenche-interns\Schematic\DFFfromNAND\DFFfromNAND.sv:1:1: , endln:10:10
  |vpiDefName:work@dff_from_nand
  |vpiName:work@dff_from_nand
  |vpiPrimitive:
  \_gate: work@nand (work@dff_from_nand.U1), line:5, parent:work@dff_from_nand
    |vpiDefName:work@nand
    |vpiName:U1
    |vpiFullName:work@dff_from_nand.U1
    |vpiPrimType:2
    |vpiPrimTerm:
    \_prim_term: , line:5:10, parent:work@dff_from_nand.U1
      |vpiDirection:2
      |vpiExpr:
      \_ref_obj: (X), line:5:10, endln:5:11
        |vpiName:X
        |vpiActual:
        \_logic_net: (work@dff_from_nand.X), parent:work@dff_from_nand
          |vpiName:X
          |vpiFullName:work@dff_from_nand.X
    |vpiPrimTerm:
    \_prim_term: , line:5:12, parent:work@dff_from_nand.U1
      |vpiDirection:1
      |vpiTermIndex:1
      |vpiExpr:
      \_ref_obj: (D), line:5:12, endln:5:13
        |vpiName:D
        |vpiActual:
        \_logic_net: (work@dff_from_nand.D), line:3:5, endln:3:6, parent:work@dff_from_nand
          |vpiName:D
          |vpiFullName:work@dff_from_nand.D
          |vpiNetType:48
    |vpiPrimTerm:
    \_prim_term: , line:5:14, parent:work@dff_from_nand.U1
      |vpiDirection:1
      |vpiTermIndex:2
      |vpiExpr:
      \_ref_obj: (CLK), line:5:14, endln:5:17
        |vpiName:CLK
        |vpiActual:
        \_logic_net: (work@dff_from_nand.CLK), line:3:7, endln:3:10, parent:work@dff_from_nand
          |vpiName:CLK
          |vpiFullName:work@dff_from_nand.CLK
          |vpiNetType:48
  |vpiNet:
  \_logic_net: (work@dff_from_nand.Q), line:2:6, endln:2:7, parent:work@dff_from_nand
    |vpiName:Q
    |vpiFullName:work@dff_from_nand.Q
    |vpiNetType:1
  |vpiNet:
  \_logic_net: (work@dff_from_nand.Q_BAR), line:2:8, endln:2:13, parent:work@dff_from_nand
    |vpiName:Q_BAR
    |vpiFullName:work@dff_from_nand.Q_BAR
    |vpiNetType:1
  |vpiNet:
  \_logic_net: (work@dff_from_nand.D), line:3:5, endln:3:6, parent:work@dff_from_nand
  |vpiNet:
  \_logic_net: (work@dff_from_nand.CLK), line:3:7, endln:3:10, parent:work@dff_from_nand
  |vpiNet:
  \_logic_net: (work@dff_from_nand.X), parent:work@dff_from_nand
```

Currently parent did not print with ref_obj and I also got nullptr in this case.

alaindargelas, 2021-06-22:
Fixed by #1409

closed 2021-06-22 by alaindargelas