chipsalliance/Surelog #1398: Param expression lacks necessary reference information

kbieganski, 2021-06-18:
In some cases, Surelog replaces parameters with expressions assigned to them. However, there are situations where it's impossible to tell what objects are being referred to.

Code:

```systemverilog
module other
  #(parameter int Width = 1)
  (input logic[Width-1:0] in, output logic[Width-1:0] out);
   assign out = in;
endmodule;

module dut (input logic[7:0] in, output logic[7:0] out);
   other #(.Width($bits({in}))) oth(.in(in), .out(out));
endmodule
```

Corresponding UHDM (with some parts cut out):

```
\_module: work@dut (work@dut) dut.v:7:1: , endln:9:10
  |vpiDefName:work@dut
  |vpiName:work@dut
  ...
  |vpiModule:
  \_module: work@other (work@dut.oth) dut.v:8: , parent:work@dut
    |vpiDefName:work@other
    |vpiName:oth
    |vpiFullName:work@dut.oth
    |vpiPort:
    \_port: (in), line:3:27, endln:3:29, parent:work@dut.oth
      |vpiName:in
      |vpiDirection:1
      |vpiHighConn:
      \_ref_obj: (in), line:8:41, endln:8:43
        |vpiName:in
        |vpiActual:
        \_logic_net: (work@dut.in), line:7:30, endln:7:32, parent:work@dut
      |vpiLowConn:
      \_ref_obj:
        |vpiActual:
        \_logic_net: (work@dut.oth.in), line:3:27, endln:3:29, parent:work@dut.oth
          |vpiName:in
          |vpiFullName:work@dut.oth.in
          |vpiNetType:36
          |vpiRange:
          \_range: , line:3:16, endln:3:25
            |vpiLeftRange:
            \_operation: , line:3:16, endln:3:21
              |vpiOpType:11
              |vpiOperand:
              \_sys_func_call: ($bits), line:3:16, endln:3:21
                |vpiName:$bits
                |vpiArgument:
                \_operation: , line:8:25, endln:8:29, parent:$bits
                  |vpiOpType:33
                  |vpiOperand:
                  \_ref_obj: (in), line:8:26, endln:8:28
                    |vpiName:in
              |vpiOperand:
              \_constant: , line:3:22, endln:3:23
                |vpiConstType:9
                |vpiDecompile:1
                |vpiSize:64
                |UINT:1
            |vpiRightRange:
            \_constant: , line:3:24, endln:3:25
              |vpiConstType:9
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
    |vpiPort:
    \_port: (out), line:3:55, endln:3:58, parent:work@dut.oth
      |vpiName:out
      |vpiDirection:2
      |vpiHighConn:
      \_ref_obj: (out), line:8:51, endln:8:54
        |vpiName:out
        |vpiActual:
        \_logic_net: (work@dut.out), line:7:52, endln:7:55, parent:work@dut
      |vpiLowConn:
      \_ref_obj:
        |vpiActual:
        \_logic_net: (work@dut.oth.out), line:3:55, endln:3:58, parent:work@dut.oth
          |vpiName:out
          |vpiFullName:work@dut.oth.out
          |vpiNetType:36
          |vpiRange:
          \_range: , line:3:44, endln:3:53
            |vpiLeftRange:
            \_operation: , line:3:44, endln:3:49
              |vpiOpType:11
              |vpiOperand:
              \_sys_func_call: ($bits), line:3:44, endln:3:49
                |vpiName:$bits
                |vpiArgument:
                \_operation: , line:8:25, endln:8:29, parent:$bits
                  |vpiOpType:33
                  |vpiOperand:
                  \_ref_obj: (in), line:8:26, endln:8:28
                    |vpiName:in
              |vpiOperand:
              \_constant: , line:3:50, endln:3:51
                |vpiConstType:9
                |vpiDecompile:1
                |vpiSize:64
                |UINT:1
            |vpiRightRange:
            \_constant: , line:3:52, endln:3:53
              |vpiConstType:9
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
    ...
    |vpiParamAssign:
    \_param_assign: , line:2:19, endln:2:28, parent:work@dut.oth
      |vpiRhs:
      \_sys_func_call: ($bits), line:8:19, endln:8:30
        |vpiName:$bits
        |vpiArgument:
        \_operation: , line:8:25, endln:8:29, parent:$bits
          |vpiOpType:33
          |vpiOperand:
          \_ref_obj: (in), line:8:26, endln:8:28, parent:$bits
            |vpiName:in
            |vpiActual:
            \_logic_net: (work@dut.in), line:7:30, endln:7:32, parent:work@dut
      |vpiLhs:
      \_parameter: (work@dut.oth.Width), line:2:19, endln:2:28, parent:work@dut.oth
        |vpiName:Width
        |vpiFullName:work@dut.oth.Width
        |UINT:1
        |vpiTypespec:
        \_int_typespec: (Width), line:2:15, endln:2:18, parent:work@dut.oth.Width
          |vpiName:Width
```

We assign the `$bits` expression to `Width`, which is used in `oth.in`'s range. In UHDM, `Width` in `oth.in`'s range is replaced with the `$bits` expression, but the `ref_obj` in it doesn't specify which `in` is being referred to. This causes an ambiguity, as it could be either `dut.in` or `oth.in`.

Even if the signal names were different, the lack of a `vpiActual` makes it difficult to figure out what the reference points to.

Note that in the `param_assign` in `dut` the `ref_obj` does have a `vpiActual` pointing to `dut.in`, which makes it unambiguous.

alaindargelas, 2021-06-19:
Short answer, it cannot be the local signal as it would be a self-reference definition, it has to be the signal in the parent module

alaindargelas, 2021-06-19:
Anyhow Surelog was not computing the actual constant value (It was not traversing the concat op and was not binding the signal properly), it does now, so it becomes a moot point.

alaindargelas, 2021-06-19:
Fixed by #1401

closed 2021-06-19 by alaindargelas